OpenTracksApp/OpenTracks #1444: New design for stickers?

dennisguse, 2023-01-08:
Back in 2019, I created a sticker for OpenTracks.
I printed two batches and those are gone now.

Question: is anybody interested creating a new design?

https://github.com/OpenTracksApp/OpenTracks/blob/main/promotion/sticker_inkscape.svg
[image]

@pstorch @rgmf @mondlicht-und-sterne

mondlicht-und-sterne, 2023-01-09:
Ich kann keine Sticker erstellen. Ich bin an neuen Stickern interessiert, für meine Ausstellungen zum ausstellen.

pstorch, 2023-01-09:
I still like the current design

dennisguse, 2023-01-22:
#1317

yuval-herman, 2023-02-03:
I am not a designer but I do know my way around inkscape. If you have any idea about a new sticker I'd more then happy to help you out!

dennisguse, 2023-02-04:
@yuval-herman not real ideas yet. 
The first sticker was inspired by these NASA mission badges and works okay as a sticker. 

As it is 3 years old already (and it took me 20min to draft the first design), I thought: can we do better?

kristicunga, 2023-02-20:
Hello 👋 

Below you can find two versions for the sticker. If you help me with some more content, I can try something better.

What do you think? 🤔 

[image]

Draft 1

[image]

Draft 2

dennisguse, 2023-02-21:
I somehow like draft two slightly more: it is better visible ;)

PS I guess, we don't need to mention Apache 2.0 anymore as it is likely not helpful (kind of too much information).

And we have an open issue for replacing privacy-proof by privacy-first (or something similar).

kristicunga, 2023-02-22:
OK, so what content do you suggest me instead of the current one?

Can we chose something of [these bullet points](https://alternativeto.net/software/opentracks/about/)? 🤔

dennisguse, 2023-04-27:
@kristicunga sorry for the real long delay.

What about these?
* A sport tracker that values your privacy
* Open-source for Android

And we could make it somehow cheesy: `with fancy features`.

MartinH80, 2023-07-02:
@kristicunga Not sure if you are still willing to finish this, please let us know.

My design skills are quite bad, but I managed to make something in inkscape that is quite similar:

[image]

I used larger fonts to make it more readable at small screens/sizes. I also put OpenTracks at the top as I expect that will be read first.

@dennisguse Please provide your feedback, next step would be my first feature branch, commit and pull request?

MartinH80, 2023-07-03:
@dennisguse Thanks :-)

I have put opentracks at the top, as that is usually where you start reading it made sense to me, but I'm not a designer.

I might need to put some more time in it and try several things like:
- some white, maybe the opentracks text?
- the privacy and open source text in smaller font size?
- rephrase to 'An open source sports tracker for Android that values your privacy'?
- do we actually need to mention that it is open source for android?
- the black circle could be smaller
- .. maybe even more.

In the beginning I thought this was a small quick fix :-)

MartinH80, 2023-07-04:
[image]
[image]
[image]

MartinH80, 2023-07-04:
@dennisguse Ok, hereby some alternatives. The longer it takes, the harder it gets :-)

dennisguse, 2023-07-06:
I think we should go with three items as more than 120° are hard to read and shorter texts (max 6 words). And in the 2nd and 3rd, the ratio between black and yellow seems to be odd. 

PS just typing on a phone in the middle of nowhere

dennisguse, 2024-02-14:
@ch3thanhs if you have fun doing graphic things: this topic is open since ages :D

ch3thanhs, 2024-02-15:
@dennisguse Sure, will try to come up with something.

ch3thanhs, 2024-02-15:
Here's my first draft, I've kept it very simple.

I'll continue to work on other variants over the weekend.

[image]

dennisguse, 2024-02-16:
@ch3thanhs you can also go creative - the sticker must no be similar to the previous one (although it may be) :D

ch3thanhs, 2024-02-17:
Draft 2 - 

[image]

ch3thanhs, 2024-02-17:
Draft 3 - 

[image]

ch3thanhs, 2024-02-18:
Draft 4 - A laptop sticker?

[image]

dennisguse, 2024-02-19:
@ch3thanhs they all look kind of great!
For no real reason, I would prefer Draft nr. 2.
How should we proceed?

@pstorch @rgmf any ideas?

pstorch, 2024-02-19:
I also like Nr. 2. Although the others are not bad either. @ch3thanhs good work.

ch3thanhs, 2024-02-23:
@dennisguse, @pstorch, Thank you!

I made some minor changes - 

[image]

Let me know if you prefer the initial version.

rgmf, 2024-02-23:
> @dennisguse, @pstorch, Thank you!
> 
> I made some minor changes -
> 
> ![One](https://private-user-images.githubusercontent.com/76928501/307399258-300c3a56-0686-4d05-93a0-77deaf61f3cd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg3MTA1MjQsIm5iZiI6MTcwODcxMDIyNCwicGF0aCI6Ii83NjkyODUwMS8zMDczOTkyNTgtMzAwYzNhNTYtMDY4Ni00ZDA1LTkzYTAtNzdkZWFmNjFmM2NkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIzVDE3NDM0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ2ZmIyNzFlYTEyZmExOTAxYzQ4MGM3ZTRjYTU1ZmM3YTNhZDE1ZGY3ZmViOTFkN2M0YmE4Mjc1NmRiNmI5NDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vaDJ89gwbsQZ-v7WVq_2BnY17VugugNOuOFeX7KMce0)
> 
> Let me know if you prefer the initial version.

I love it :heart:

dennisguse, 2024-02-26:
@ch3thanhs that looks great!
Can you create a PR for logo nr. 3?
I guess, it goes into the `promotion`-folder (and we need to move the other files around ;)
And we can also replace the feature graphic for F-Droid: `fastlane/metadata/android/en-US/images/featureGraphic.png`
This one actually never looked great in F-Droid....

And I would then try to figure out how to get some stickers printed aka a PDF with the correct information how to cut the stickers...

dennisguse, 2024-02-28:
Closed by 6657eca54d3aecd3daeb862877cc2dabddfc0e0b

closed 2024-02-28 by dennisguse

dennisguse, 2024-07-02:
@ch3thanhs I just got some printed stickers :D 
Great work!

[image]