OpenTracksApp/OpenTracks #1448: Bluetooth sensor statistics missing from recorded tracks in v4.2.0

Salvoxia, 2023-01-18:
Hi, 

I use OpenTracks for recording my cycling sessions with bluetooth speed-, cadence- and power-meters connected.
After updating to v4.2.0 (F-Droid) average/max cadence and average power are now missing when displaying old recorded sessions. The data is still there in the diagrams, but no longer in the stats (first page).
Am I missing something or is this a bug?

Best Regards,
Salvoxia

v4.1.2 vs v4.2.0
[image] [image]

dennisguse, 2023-01-18:
Good question. 
I have to check....

dennisguse, 2023-01-19:
Found it: functionality was removed and needs to be re-implemented due to some technical changes for other UI features.

Here is the issue for this: #1426

closed 2023-01-19 by dennisguse