OpenTracksApp/OpenTracks #1452: Add 2 workouts without GPS, with the ability to edit the distance of the workout.

MrS1R1US, 2023-01-25:
Since it is winter now, in Ukraine with war there and some difficulties with electricity and street lighting, please add 2 types of training: Elliptical trainer and Treadmill. I'll be very thankful.

Also, if possible, please add the ability to edit the distance to specify it from the bike path or elliptical trainer.

I'm adding large icons, maybe someone can add a person to the Elliptical trainer:
[image]
[image]

dennisguse, 2023-01-26:
How would you call these two sports (in English)?
I can add those already without the icons.

PS/ entering distance manually; see #937.
This is requires quite some refactoring.

MrS1R1US, 2023-01-26:
> How would you call these two sports (in English)? I can add those already without the icons.
> 
> PS/ entering distance manually; see #937. This is requires quite some refactoring.

"Elliptical Workout"
and 
"Treadmill workout"

You can temporarily use these icons, one of which will be finalized and replaced. The second icon should suit the style of the application.

Thank you.

dennisguse, 2023-05-23:
@MrS1R1US Are the icons ready to use?

closed 2023-09-27 by dennisguse