OpenTracksApp/OpenTracks #1454: Time tracking sometimes loops back

ScriptForAll, 2023-01-25:
**Describe the bug**
A clear and concise description of what the bug is.

I noticed this bug because my phone constantly loses its GPS signal. When I lose GPS signal and then it's back up again the time goes back to when the GPS signal was lost.
However, this bug doesn't happen every time I lose signal only sometimes. 
I've also noticed that the "Moving Time" keeps tracking despite not having the GPS signal.

**To Reproduce**
1. Start an activity
2. GPS Signal lost
3. Moving and Total Time keeps tracking
4. GPS signal restored
5. Both Moving and Total Time go back to the time when the signal was lost

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: Redmi Note 7
 - OS: MIUI Global 11.0.6
 - OpenTracks version: v4.2.0 (F-Droid)
 - OpenTracks commit id (only for nightly builds): N/A

dennisguse, 2023-01-25:
Interesting - could you make a screen recording when this appears?

And we have a problem with the GPS signal lost warning: #1429.

All of this should  not affect actually recorded data.

ScriptForAll, 2023-01-25:
Once I get to recreate this, of cource! :+1:

dennisguse, 2023-02-18:
Can confirm: I have seen this in the wild.
Moving time keeps going while there is no GPS. 

Also total time does not account for duration of lost GPS. 

Problem is likely in TrackStatisticsUpdater.

ScriptForAll, 2023-03-01:
Hi!
I couldn't document this  in video after all because my smartphone is now broken.
However, I know now a way to recreate this bug in viable manner.

My method is, while driving, to cross a tunnel that's deep enough to not have GPS signal inside of it. Whenever the GPS signal is restored, the time loops back to the "timestamp" of when the app warns about losing the GPS signal.

> Moving time keeps going while there is no GPS.
> Also total time does not account for duration of lost GPS.

After doing this for a while I can confirm say that what you say here is true although sometimes "Total Time" keeps also going.

dennisguse, 2023-03-25:
So, I was just looking into this and it is tricky.
we might be able to do something about this, but may not be able to avoid this completely.

First thing: what is `total time`?
`total time` in OpenTracks is the duration of all segments so far.
The duration of a segment is `stopTime - startTime`.
Segments get created if two subsequent GPS locations are two far apart (default 200m).
If there is a gap between two segments, this will also include some duration.
And this will not be counted to `total time`.

Lets assume we are in a track with one segment.
When GPS is lost, the UI will be updated every second still.
Let's assume we are not receiving new GPS data while we are still moving and at some point in time we will have crossed this new-segment 200m boundary.
Now, `total time` would still increase as no new GPS was received...
When we then receive a GPS location, a new segment is created and `total time` recomputed; thus jumping back as the time between segments is not counted.

@pstorch @rgmf any ideas about total time?
Does total time how OpenTracks does it make sense or should it rather be duration since recording started?

PS/ same applies to `moving time`, but we can do something about this - we count everything to moving time if it has a speed greater than a threshold. For UI updates, we use the speed of the last GPS-based trackpoint (if no speed sensor is presented). And this GPS speed has no timeout. We could set this to 5s (similar to speed sensor) and the moving time would **jump** less backward.

rgmf, 2023-03-25:
I can't remember that part of the code but thinking about this:
- Total time should be stoptime - starttime, regardless segments and so on. It should be as simple as that, right?
- About moving time, it's difficult because the segments logic you point out and the GPS signal issues.

I've hardly ever had this kind of issues. In fact I can't remember if I got this issue.

dennisguse, 2023-03-25:
Currently `Total time = t3 - t0 - (t2 -t1)` or` (t1 - t0) + (t3 - t2)`
Could it be that this is due to pause/continue functionality?
If I use this functionality, I wouldn't expect the time to show up in the `total time`, right?
Something like I start recording in the morning, then pause for lunch break, and continue afterwards.
Then the lunch break would not count to `total time`.

So, for "too much distance"-segments, we could do t3-t0? And for the manually created segments, we can use the "old" algorithm.
I guess, we can use the SEGMENT_MANUAL TrackPoints to distinguish these two cases.

Moving time depends if consecutive points are considered moving; within a segment there are moving TrackPoints and non-moving TrackPoints.

rgmf, 2023-03-25:
I'd count lunch time as part as the activity total time. It has sense for me 🤔

pstorch, 2023-03-25:
Ok, I understand the question, if the time between two segments should be counted on the total time. Hmm, maybe depending on the user activity. If you have a whole day hiking or biking with a bigger lunch break. Then I was a full day on tour but inactive in between. Hmm, I don't know. Maybe a user choice. On the other hand one can also record two activities, before and after lunch, if the break shouldn't be counted. 

Back to the time loop issue. I still don't get the situation when this can happen.

dennisguse, 2023-03-25:
It happens, because data from GPS has no TTL. 

GPS data is received (moving / with speed) and then no further data is received. 
However, we cache this TrackPoint to show it's speed, coordinates and altitude in the UI. 
And when the every second UI updates gets the TrackPoint and uses it to generate a TrackStatistic for the UI. As the GPS data is suggesting movement, moving time is counted. 

To fix this, we have to forget the GPS data if it gets too old - then the UI update will not considered be moving.

pstorch, 2023-03-26:
Ok, I think I got it now.
Maybe total time should not be affected by GPS lost. Only for manual pause/stop (for a lunch break :wink:).
Moving time: either stop the moving time after "reuse" of the last trackpoint for 2-3 seconds. Or indicate the possible wrong moving time by making it red and blinking. Or both.

pstorch, 2023-03-27:
Tested yesterday again and observed the moving time. Never did that before.
It stops when no move is detected and continues when moving again. That's fine. I'm more and more convinced that the moving time should stop when no GPS signal is received and not pretend the moving continues. That might result in a jump forward when new valid GPS trackpoints are received, but better this way than pretending a wrong movement.

dennisguse, 2023-03-27:
:+1: 

What about the following plan?
1. We limit the TTL of GPS data to 5s
2. Time between segments (aka not SEGMENT_MANUAL) is counted to `total time`
3. (optional) for stale coordinates we add some visualization (like color)

dennisguse, 2023-09-27:
As #1187 (idle detection due to timeout), this shouldn't be a problem anymore.