OpenTracksApp/OpenTracks #1457: Missing heart rate history

MrS1R1US, 2023-01-29:
**Describe the bug**
After recording a workout, looking at the current and previous ones, the average and maximum heart rate indicators are not shown, as it was before. I don't know at what point it happened because I didn't train for several months. The last time I saw this data was 29.10.22. HELP, This is very important data for the analysis of the current and past training.

**To Reproduce**
1. Go to 'any recorded workout with heart rate'
2. See error

**### OLD:**
[image]

**### NOW:**
[image]

**Technical information**
 - Device: Asus ZenFone Max Pro M2
 - OS: LineageOS 16.0
 - OpenTracks version: v4.2.1

dennisguse, 2023-01-29:
This feature was removed in v4.2.0 due to some ugly but required refactoring (mainly to implement heart rate zones).
Downgrading to v4.1.0 is the only option to get this feature back for now.

https://github.com/OpenTracksApp/OpenTracks/issues/1426#issue-1459138309

closed 2023-01-29 by dennisguse

MrS1R1US, 2023-01-29:
> This feature was removed in v4.2.0 due to some ugly but required refactoring (mainly to implement heart rate zones). Downgrading to v4.1.0 is the only option to get this feature back for now.

What for? She worked just fine...

When will it be restored? At least approximately

dennisguse, 2023-01-29:
It was working fine, but made changes very complicated.
Now it is easier (more flexible), but this feature is gone now.
From my side: not before May.
Somebody else might pick it up though.

MrS1R1US, 2023-01-29:
> It was working fine, but made changes very complicated. Now it is easier (more flexible), but this feature is gone now. From my side: not before May. Somebody else might pick it up though.

Please don't forget this feature, it's important. I'll be waiting. Unfortunately, so far on the version below (

MrS1R1US, 2023-01-30:
@dennisguse Tell me, how can I understand when these heart rate numbers will return? After all, I had to downgrade and not accept any updates now.

MrS1R1US, 2023-03-21:
> This feature was removed in v4.2.0 due to some ugly but required refactoring (mainly to implement heart rate zones).
> Downgrading to v4.1.0 is the only option to get this feature back for now.

Where can I download apk file version 4.1?

dennisguse, 2023-03-21:
@MrS1R1US you need to compile it yourself. 
It isnot on F-Droid anymore. What happened?

PS might be still available via Google Play Store.

MrS1R1US, 2023-03-21:
> @MrS1R1US you need to compile it yourself. It isnot on F-Droid anymore. What happened?
> 
> PS might be still available via Google Play Store.

I need the average result of the heart rate sensor at the end of the workout. I could not connect the sensor and assumed that the problem was in the version, but it turned out that it did not search because the GPS was turned off. Now I need to go back to version 4.1, otherwise, without an average result, it makes no sense for me to use the application (((

MrS1R1US, 2023-03-24:
> @MrS1R1US here is v4.1.2 PlayStore build. [OpenTracks v4.1.2.zip](https://github.com/OpenTracksApp/OpenTracks/files/11033288/OpenTracks.v4.1.2.zip)
> 
> And I learned that I can download old versions from the PlayStore sunglasses

Thank you very much. I hope this bug will be removed soon, and the HR story will be the same.