OpenTracksApp/OpenTracks #1465: STATS page bpm field shows sensor name+MAC, not heart rate

awm0, 2023-02-03:
[image]
Version 4.2.1 5132, all prevous versions I used since end 2021 worked fine

dennisguse, 2023-02-04:
Already fixed; see #1455. 
And available in the nightly. 

@rgmf @pstorch Could on of you create a release?
I don't have a computer at hand in the next days.
That would be great 😎

pstorch, 2023-02-04:
Not done it before, but the documented release process looks familiar. I'll give it a try.

dennisguse, 2023-02-04:
@pstorch looks great!
Thanks :)

closed 2023-02-04 by dennisguse

pstorch, 2023-02-04:
Release [v4.2.4](https://github.com/OpenTracksApp/OpenTracks/releases/tag/v4.2.2) is tagged and on it's way to F-Droid (hopefully).