OpenTracksApp/OpenTracks #1466: .gpx only shows heartrate dot in graph when there is simultaneous GPS coordinates after import into GC

awm0, 2023-02-03:
OpenTracks 4.2.1 5132, Golden Cheetah 3.5 Appimage
Steps for Reproduction:
Record indoor Activity (e.g. rowing) in OpenTracks with GPS activated (don't know how to deactivate for indoor)
using Polar H10 belt
Tab "Plot by time" shows continuous heart rate graph throughout activity
[image]
Export to .gpx
Import .gpx to GoldenCheetah
[image]
The heart rate graph (red) shows wide gaps
Export to .csv
(Import into GC not possible, no valid data error)
[2023-01-07_11_58_59.508_2023-01-07T11_58Z.csv](https://github.com/OpenTracksApp/OpenTracks/files/10583542/2023-01-07_11_58_59.508_2023-01-07T11_58Z.csv)
Spreadsheet in LibreOffice shows that heartrate values are present throughout the recording,
The periods with plotted heartrate graph in GC correllate with the presence of GPS coordinates in .csv (and .gpx?!)

dennisguse, 2023-02-04:
Sadly, the GPX standard doesn't support data without coordinates.
So, OpenTracks ignores the intermediate data points if no coordinates are present. 

This is support by KML though and OpenTracks exports all data points.

dennisguse, 2023-02-04:
IIRC GoldenChetah doesn't support KML import, right?

closed 2023-02-04 by dennisguse