OpenTracksApp/OpenTracks #1492: Support for Androids new AltitudeConverter

dennisguse, 2023-02-21:
**Is your feature request related to a problem? Please describe.**
Android U will provide the functionality to convert WGS84 to altitude above sea level (TBD which model).

**Describe the solution you'd like**
Support both EGM2008 as well as Androids new AltitudeConverter (if present).

I guess, we should use the internal EGM2008 only as fallback.

https://developer.android.com/reference/android/location/altitude/AltitudeConverter

**Additional context**

https://issuetracker.google.com/issues/195660815#comment1

gdt, 2023-02-21:
Is that an AOSP API, or a proprietary API?   I wonder if when targetting an older API there will be a fallback in the build system?  Surely it doesn't make sense to require a high minSKD, and is there a way to check/use at runtime?

dennisguse, 2023-02-21:
It should be AOSP and raising minSDK is not required (OpenTracks is still on Android 5.1). 
We could just do the check at runtime and use the new API if it is present. 

In the long run, we could have different build flavors.

commit 8c364ba7 (2023-06-22): Support altitude correction via Android's native AltitudeConverter (API34+).
Fixes #1492.
closed 2023-06-22 by dennisguse