OpenTracksApp/OpenTracks #1493: UI: search's close icon might be overlapped

dennisguse, 2023-02-27:
**Describe the bug**
In the tracklist, the close button from the search may be overlapped by the record start button. 

**To Reproduce**
1. Go to Track Lidt
2. Click on search icon in action bar 
3. Depending on the screen width the close icon for the search might be overlapped

**Technical information**
 - OpenTracks version: v4.3.2
[image]

rgmf, 2023-02-27:
Good catch!! It seems it only happens in some phones, so I guess it depends on the resolution... so we have to think a better solution.

dennisguse, 2023-02-27:
Wasn't me ;)
I got the report via email same days ago....

PR #1597 (2023-06-15): Migrate to Material3 design.
Fixes #1586.

Should work with Material3's DynamicColor (cannot test due to lack of support on my device).

Changes: 
* re-introduced topbar for all activities excl. TrackRecording and TrackRecorded
* TrackList split topbar menu and bottombar menu
* moved search (TrackList and MarkerList) to topbar (fixed #1493)
* tracklist now shows search query in topbar:  part of #1545

Non-change:
* Record button remains on Material2 as it is round while Material3 not

[image]

[image]
[image]
closed 2023-06-10 by dennisguse