OpenTracksApp/OpenTracks #1496: Android 14 changes

dennisguse, 2023-03-06:
Android 14 is in preparation and some of the changes will require changes in OpenTracks: https://developer.android.com/about/versions/14/behavior-changes-14

* [ ] ForegroundServices must provide service type
* [ ] TargetVersion update (build.gradle)
* [ ] Java 17 becomes available?

dennisguse, 2023-04-13:
@pstorch It could be that I broke yesterday the nightly builds.
I upgraded from JDK11 to JDK17 and I guess this is not installed on the build machine, right?

pstorch, 2023-04-13:
Yeah, looks like: https://fdroid.storchp.de/fdroid/repo/build.log

I'll try to fix it.

pstorch, 2023-04-13:
@dennisguse do you know if the official fdroid buildserver supports it?

dennisguse, 2023-04-13:
@pstorch supported yes, activated no. For JDK11, we added some manual installation steps in the fdroid config. 
IIRC it is a debian machine and this openjdk 17 should be there. 
Thanks for reminding me :)

pstorch, 2023-04-13:
I installed openjdk-17-jdk on the build server but now I get a different error:

```
##################
# build OpenTracks
##################
Already on 'main'
Your branch is up to date with 'origin/main'.
Already up to date.
last commit 7d58014b1, new commit 7d58014b1, commit count 5175
Repository changed, updating
Downloading https://services.gradle.org/distributions/gradle-8.0.2-bin.zip
...........10%............20%............30%............40%............50%............60%...........70%............80%............90%............100%

Welcome to Gradle 8.0.2!

Here are the highlights of this release:
 - Improvements to the Kotlin DSL
 - Fine-grained parallelism from the first build with configuration cache
 - Configurable Gradle user home cache cleanup

For more details see https://docs.gradle.org/8.0.2/release-notes.html

To honour the JVM settings for this build a single-use Daemon process will be forked. See https://docs.gradle.org/8.0.2/userguide/gradle_daemon.html#sec:disabling_the_daemon.
Daemon will be stopped at the end of the build 
Checking the license for package Android SDK Build-Tools 33.0.1 in /opt/android-sdk-linux/licenses
License for package Android SDK Build-Tools 33.0.1 accepted.
Preparing "Install Android SDK Build-Tools 33.0.1 v.33.0.1".
"Install Android SDK Build-Tools 33.0.1 v.33.0.1" ready.
Installing Android SDK Build-Tools 33.0.1 in /opt/android-sdk-linux/build-tools/33.0.1
"Install Android SDK Build-Tools 33.0.1 v.33.0.1" complete.
"Install Android SDK Build-Tools 33.0.1 v.33.0.1" finished.
Checking the license for package Android SDK Platform 33 in /opt/android-sdk-linux/licenses
License for package Android SDK Platform 33 accepted.
Preparing "Install Android SDK Platform 33 (revision 2)".
"Install Android SDK Platform 33 (revision 2)" ready.
Installing Android SDK Platform 33 in /opt/android-sdk-linux/platforms/android-33
"Install Android SDK Platform 33 (revision 2)" complete.
"Install Android SDK Platform 33 (revision 2)" finished.
> Task :compileNightlyJavaWithJavac FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileNightlyJavaWithJavac'.
> Could not resolve all files for configuration ':androidJdkImage'.
   > Failed to transform core-for-system-modules.jar to match attributes {artifactType=_internal_android_jdk_image, org.gradle.libraryelements=jar, org.gradle.usage=java-runtime}.
      > Execution failed for JdkImageTransform: /opt/android-sdk-linux/platforms/android-33/core-for-system-modules.jar.
         > Error while executing process /usr/lib/jvm/java-17-openjdk-amd64/bin/jlink with arguments {--version}

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 4s
19 actionable tasks: 1 executed, 18 up-to-date
sending incremental file list

sent 7,745 bytes  received 70 bytes  15,630.00 bytes/sec
total size is 4,913,262  speedup is 628.70

```

pstorch, 2023-04-13:
Now it worked

dennisguse, 2023-04-13:
@pstorch why?

pstorch, 2023-04-13:
Actually I don't know. I saw some other issues and rebooted the VM. After that also the build with Java 17 worked 🤷‍♂️
"Ein Boot tut gut" 😁

commit 81b09089 (2023-06-08): Target Android 14/API34.

Fixes #1496.
closed 2023-06-08 by dennisguse