OpenTracksApp/OpenTracks #1510: Automatically activate the GPS when opening the app

MrRar, 2023-03-29:
Currently starting recording is a two step process. First enable the GPS in the app and then wait for it to lock and then press the record button.

Enable the GPS automatically when the app is opened. I don't see the point of recording without the GPS. There's already an app for that, Clock. I think it would be best to enable the GPS whenever the app is opened by default but at the very least there should be a setting to do that.

dennisguse, 2023-03-29:
Actually, you don't have to start GPS - it is happening automatically.
It is for people where the time to GPS fix (like older phones) is too long and should not be recorded.

MrRar, 2023-03-29:
> Actually, you don't have to start GPS - it is happening automatically. It is for people where the time to GPS fix (like older phones) is too long and should not be recorded.

My phone isn't really to old (samsung j3 orbit) but it is a cheep one. It usually takes a few minutes for the GPS to lock on my phone. I didn't know that is abnormal. This may not be as big a problem as I thought.

The app only tries to lock the GPS when a recording is started. It would be nice for people like me to have a setting to get the app to start trying to lock the GPS as soon as the app is opened. Then all I need to do is wait for it to lock and then start recording.

closed 2024-12-29 by dennisguse