OpenTracksApp/OpenTracks #1512: Bluetooth sensors broken after updating to Android 13

ae5960e8-a6fc-491f-b252-898ecf59af95, 2023-04-03:
**Describe the bug**
I can choose my Bluetooth devices from the OpenTrack settings but no input is recorded or shown.

**To Reproduce**
Android 13, Bluetooth speed sensor DuoTrap S.

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: Note 10 iodeOS
 - OS: 4.1 
 - OpenTracks version: 4.3.0

dennisguse, 2023-04-03:
Mhhh. 
Hypothesis: There might be a missing/not granted permission. 

1. Can you check that Bluetooth permissions are granted?

2. Can you remove all permissions for OpenTracks (set them to ask everytime)?

ae5960e8-a6fc-491f-b252-898ecf59af95, 2023-04-03:
Confirmed issue is not with OpenTracks.

closed 2023-04-03 by ae5960e8-a6fc-491f-b252-898ecf59af95

dennisguse, 2023-04-03:
@ae5960e8-a6fc-491f-b252-898ecf59af95 how did you solve it?

dennisguse, 2023-04-03:
@ae5960e8-a6fc-491f-b252-898ecf59af95 and can you add the sensor (as it is working) to the DEVICES.md?

ae5960e8-a6fc-491f-b252-898ecf59af95, 2023-04-03:
Haven't solved it, the issue is with the new version of the OS my phone is on, so the issue isn't with the app.