OpenTracksApp/OpenTracks #1522: Generated KML files are malformed

julien-nc, 2023-04-12:
Thanks for this amazing app :blue_heart: 

Here are the first few lines of a KML file (or a KML file inside a KMZ) generated by OpenTracks :
```
<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns="http://www.opengis.net/kml/2.3"
xmlns:atom="http://www.w3.org/2005/Atom"
xmlns:opentracks="http://opentracksapp.com/xmlschemas/v1">
xsi:schemaLocation="http://www.opengis.net/kml/2.3 http://schemas.opengis.net/kml/2.3/ogckml23.xsd http://opentracksapp.com/xmlschemas/v1 http://opentracksapp.com/xmlschemas/OpenTracks_v1.xsd">
<Document>
```
The <kml> tag is closed and then the `xsi:schemaLocation` attribute is added.

This is most likely the cause of https://github.com/OpenTracksApp/OpenTracks/issues/1388 as the linked kml file ([doc.kml.txt](https://github.com/OpenTracksApp/OpenTracks/files/9713511/doc.kml.txt)) is malformed in the same fashion.

**To Reproduce**
1. Choose KML or KMZ as export format in OpenTrack settings
2. Export a track

**Technical information**
 - OS: LineageOS 20.0
 - OpenTracks version: v4.3.0

dennisguse, 2023-04-12:
Confirmed by reading the code.
Introduced in: 82fe859c846b2dc96b8980990c8d71482bc2413f

closed 2023-04-13 by dennisguse

dennisguse, 2023-04-13:
Thanks for the report! :+1:

commit 93bcf8d4 (2023-04-13): Bugfix: KML header malformed.
Fixes #1522.
Introduced in 82fe859c846b2dc96b8980990c8d71482bc2413f