OpenTracksApp/OpenTracks #1527: UI shows incorrect language

nthtimes, 2023-04-14:
UI shows incorrect language in some cases like the following:
- zh-TW: It shows the en language
- zh-Hk: It shows the zh language

**Technical information**
 - Device: ASUS Zenfone 6 (2019)
 - OS: OmniROM 13
 - OpenTracks version: v4.3.0, v4.3.1

dennisguse, 2023-04-14:
IIRC that is the default behavior for missing translations in the regional specific locales. 
First check zh-tw, then zh and then fallback to default  (on Android this is en-us).

nthtimes, 2023-04-14:
I note an untranslatable string named settings_layout_reset_key was translated in values-zh-rTW/strings.xml.
Is this the reason?

dennisguse, 2023-04-15:
Actually, those should not be translated as they are purely technical. I will remove these translations.

dennisguse, 2023-04-18:
Since you are on Android 13, how do you set the language?

PS/ `settings_layout_reset_key` is now in the correct file and cannot be translated anymore.

nthtimes, 2023-04-18:
To set the language as zh-TW:
1. Going into "App info\Language" on OpenTracks.
2. Choose "中文 (台灣)", that is "Chinese (Taiwan)"

nthtimes, 2023-04-18:
This issue started in v4.3.0.
In the old version 4.2.2, I got the correct language that same the system setting.

dennisguse, 2023-04-18:
@elastic-rock this seems to be an effect of #1468.
Any ideas?

dennisguse, 2023-04-24:
@elastic-rock ping

elastic-rock, 2023-04-24:
I confirm that I am alive and know about it, but I unfortunately had not the time to do it yet. I shall do it in a week max.

elastic-rock, 2023-04-29:
https://github.com/OpenTracksApp/OpenTracks/pull/1540#issuecomment-1528784034

dennisguse, 2023-05-01:
@nthtimes could you check that this fixing your issue?
Nightly is already available and F-Droid might take some days as usual.

nthtimes, 2023-05-01:
It is correctly shows my language now.
But I found that all the regional option are getting the same translation, the Traditional Chinese (Hong Kong) seems to show zh-TW translation.

dennisguse, 2023-05-01:
So, now it gets dirty.

I did some digging into when things started going wrong.

I compiled the APK for different versions and checked the included locales `resources.asc/strings`
* v4.3.4 does include 44 locales (no country-specific locales)
* v4.3.0 does include 44 locales (no country-specific locales)
* v4.2.2 does include 92 locales (everything is included)

In the repo, we have 54 locales...

So, all the configuration for Android 13's per app language support (included in v4.3.0) some leads to omit locales (especially country-specific locales).
This is not fixed by autogenerating these config files as it is done in v4.3.4 (Android build tools 8.1.0).

dennisguse, 2023-05-01:
... but all that does not explain why the wrong regional locale is shown now.

dennisguse, 2023-05-16:
@nthtimes could you check if this is still broken for you using the v4.3.5?

dennisguse, 2023-05-19:
I just tested this using an Android 13 emulator and replaced zh, zh_HK, and zh_TW with a locale I can distinguish.
When doing this I get three different localizations.

The regions I get shown are these (screenshot below).
I translated the screenshot using Google Translate and all looks good.

Don't really know what else to test :/

[image]

nthtimes, 2023-05-19:
[image]
[image]
[image]

After changing into zh_HK traslication, it seems good, but not.
I think I got the zh_tw translation, because the zh_hk translation has no any the translated strings in About OpenTracks.

dennisguse, 2023-05-20:
In Weblate, we have three flavors of Chinese:
* https://hosted.weblate.org/projects/opentracks/opentracks-src-main-res-values-strings-xml-main/zh_Hant/#information mapped to `zh`
* https://hosted.weblate.org/projects/opentracks/opentracks-src-main-res-values-strings-xml-main/zh_Hant_HK/#information mapped to `zh-rHK`
* https://hosted.weblate.org/projects/opentracks/opentracks-src-main-res-values-strings-xml-main/zh_Hans/#information mapped to `zh-rTW` (Taiwan)

First, I don't know why Weblate is mapping zh_Hans to zh-rTW.
Second, Android will use `zh` aka zh_Hant as fallback if a translation is `zh-rHK` is missing.
Some more infos here: https://developer.android.com/guide/topics/resources/multilingual-support

I am really at a loss here on how to continue.

commit 63c7ecf3 (2023-05-01): Build: remove resConfigs limiting included languages.

Fixes #1527.
commit dcad42bf (2023-05-01): Build: remove resConfigs limiting included languages.

Fixes #1527.
closed 2023-05-25 by dennisguse