OpenTracksApp/OpenTracks #1532: 2 tracks same device different results

nutpantz, 2023-04-16:
>two tracks both recorded at the same time on the same device different results. v 3.7.4 and latest nightly 4.3.1-12 gf.

If both apps are running at the same time on the same device with the same settings should they not have the same results? And the exact same track? But one is 9.06km and the other is 8.28km

V3.74 does not have the gps disabled flashing error which is why i run it.

dennisguse, 2023-04-17:
Which one do you expect to be correct?
When did this behavior start?

dennisguse, 2023-04-17:
What happens if you import the v3.7.4 generated file into v4.3.1?

dennisguse, 2023-04-17:
v3.7.4: has 2553 trackpoints / "<gx:coords>
v4.3.1: has 1939 locations / "<coords>"

Please note: more trackpoints does not imply "better data".

And v3.15.0 had a serious change: https://github.com/OpenTracksApp/OpenTracks/releases/tag/v3.15.0
@nutpantz could you try this as well in parallel to v3.7.4?

nutpantz, 2023-04-18:
If there was some where i could get the 3.15 apk (or the 3.15 nightly apk) i could try it with the latest nightly.
(I can run a nightly and release at the same time)

dennisguse, 2023-04-25:
@nutpantz ping

nutpantz, 2023-04-25:
Blew out my back, can't test right now

dennisguse, 2023-04-26:
@nutpantz autsch!
Get well soon!

nutpantz, 2023-04-27:
[kml2.zip](https://github.com/OpenTracksApp/OpenTracks/files/11347646/kml2.zip)

Short track small differences.

nutpantz, 2023-05-04:
4 tracks all on the same device at the same time 3 open track vers and one other app. 
Other app and 3.?? Record 9.?? Km 
2 vers of opentrack 4.?? Record 7.?? Km.   
  
I would think something is wrong.

nutpantz, 2023-05-14:
[more tracks.zip](https://github.com/OpenTracksApp/OpenTracks/files/11470966/more.tracks.zip)
And again 15k trip the newest app is almost 1km less then 3.15 and a different app all recorded on the same device at the same time. (Also on another device that was also running same results)

dennisguse, 2023-05-23:
I doubt it, but #1566 might be related.

dennisguse, 2023-09-27:
Hopefully, this isn't a problem anymore.

closed 2023-09-27 by dennisguse