OpenTracksApp/OpenTracks #1541: Record only on gyro activity

Bernie-KN, 2023-04-19:
Often my hiking tours are interrupted by a few drinks, sandwiches and photos (digicam). Although inactivity and accuracy parameters are set, I get spikes (waypoints) up to 100m around my true location, which unnecessarily lengthen my recorded routes.

Would it be possible to implement an option that only records when gyroscope sensor activity is detected (with adjustable threshold)?

dennisguse, 2023-04-19:
Like #30?
I never had the time to build this, but it is doable (if somebody) has the time...

dennisguse, 2023-08-08:
Will not be necessary after implementing idle detection via timeout.
We can make the distance before switching from idle to non-idle configurable.

See #1187

closed 2023-08-08 by dennisguse