OpenTracksApp/OpenTracks #1542: Import/Export to Health Connect

thomassth, 2023-04-20:
**Is your feature request related to a problem? Please describe.**

It'd be very convenient to retrieve / record data to and from Google fit, Samsung health and other apps, potentially opening data retrieval from smartwatches and other proprietary devices, or incorporate new optional data types.

Depend on how the API is used, we may get to selectively export / import data

https://developer.android.com/guide/health-and-fitness/health-connect/get-started

dennisguse, 2023-04-20:
What is your problem that this would be solving?

dennisguse, 2023-04-24:
Is the GPX/KML import not sufficient?

thomassth, 2023-04-25:
Current versions of Google fit lacks any kind of file export

dennisguse, 2023-04-25:
We had a similar question regarding upload to Google Fit: https://github.com/OpenTracksApp/OpenTracks/discussions/1152

You can manually export from Google Fit and then check if you can import this data into OpenTracks: https://support.google.com/fit/answer/3024190?hl=en

A direct integration requires including proprietary libraries which is a show stopper.
Also it likely requires Internet access, which is another show stopper.

closed 2023-04-25 by dennisguse

thomassth, 2023-04-25:
I don't know where you're getting that showstopper conclusion, as health connect works on a device level and does not require internet permission.

dennisguse, 2023-04-26:
1. Google Fit Android API requires PlayServices
2. Google Fit Android API license is not obvious (likely cannot be included in F-Droid builds)
3. Google Fit Android API is deprecated (EOL 2024): https://developers.google.com/fit/android
4. Google Fit REST API requires Internet access

Same applies to successor Health Connect: https://developer.android.com/guide/health-and-fitness/health-connect/platform-overview/availability

Don't get me wrong: it would be doable (if somebody wants to put in the work), but is likely very unhealthy for OpenTracks.
If you want to build this: create a separate application that fetches the data from Google Fit, converts it into a file formation supported by OpenTracks and trigger the import.

aveao, 2025-05-09:
@dennisguse as of Android 14, health connect is built into AOSP. As such it does not require play services and to my understanding the API license should be F-droid compatible.

https://developer.android.com/health-and-fitness/guides/health-connect/develop/get-started

Would you be open to reconsidering opening this issue or would you be open to another issue on health connect support, especially for exporting fitness data?

dennisguse, 2025-05-09:
@aveao Sure. 
I don't know exactly the use case besides storing data in a central (on-device) database. 

What would be your use case?

aveao, 2025-05-09:
I'm interested in the ability to access and view historic fitness tracking and vitals data (heartrate, sleep tracking etc) from different sources in one place.

I must admit I do plan on using google fit for the other half of this equation, but adding support for this would also allow other apps to be able to provide the same functionality, provided someone puts in the work to write it.

Exporting data regularly to import it elsewhere is a viable option (which the app currently supports), but something automated with a standardized API would of course work better.

dennisguse, 2025-05-09:
Sounds reasonable. 

My usual response: if somebody wants to implement this and it makes sense - go for it.