OpenTracksApp/OpenTracks #1546: Total Altimeter Support

SiriusLightning21, 2023-04-24:
I use Open Tracks for Mountainbiking and hiking. Is it possible to add the total altimeter like on strava? Is there an other app that can do it with the data from Open Tracks?

dennisguse, 2023-04-24:
Based upon which data/sensor?

OpenTracks supports use of a barometer (if present) to measure gain/loss via air pressure.

AAT is using an alternative approach: it uses measured altitude data to derive the gain/loss from the traveled path.
https://bailu.ch/aat/

SiriusLightning21, 2023-04-24:
> Based upon which data/sensor?
> 
> OpenTracks supports use of a barometer (if present) to measure gain/loss via air pressure.
> 
> AAT is using an alternative approach: it uses measured altitude data to derive the gain/loss from the traveled path.
> https://bailu.ch/aat/

Based from my Smartphone data. When I click in Open Tracks "Nach Zeit" After Time I see the heighs data at diffrent times from my track. But not the total heighs from the track like on Strava or other tracking apps for Mountains.

dennisguse, 2023-04-24:
If you have a barometer, the gain/loss will be shown on the first page (statistics) and the 2nd one (interval; here per interval).

dennisguse, 2023-04-24:
See #1619.

closed 2023-04-24 by dennisguse