OpenTracksApp/OpenTracks #1548: Using the code

ugintl, 2023-04-24:
Can I upload the app to my play store with my own name?

ugintl, 2023-04-25:
Yes I will modify it

pstorch, 2023-04-25:
@ugintl you will also do support for your fork then?
Beware that you might have to justify against Google, if your app is too similar. They check for duplicates and rip offs.

ugintl, 2023-04-25:
I will make sure my app is different than yours. While checking for duplicates what Google focuses on?

dennisguse, 2023-04-25:
@ugintl what would you like to change?
And why can this not be integrated into OpenTracks?

Are you planning a hard fork or do you plan to pull changes from OpenTracks in the future?

Who is your user base?

dennisguse, 2023-04-27:
@ugintl that does not really answer the question, why would want to you want to do :man_shrugging: 

Anyhow:
* Code is open source (Apache license) and you have to adhere to it
* You have to use a different logo (otherwise might be flagged later by PlayStore)
* You should use a different application name
* You should make clear that is NOT OpenTracks (and note somewhere visible for the user a fork)

To be honest, I still didn't understand why you want to do this.
Please keep in mind that creating a fork (which this is likely going to be), might have some severe side effects.
Some discussions here: 
https://softwareengineering.stackexchange.com/a/253940

closed 2023-04-30 by dennisguse