OpenTracksApp/OpenTracks #1551: Miband using gadgetbridge

Dambond, 2023-05-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
1. Go to start sport on the band

2. Click on finish the sport in th band

3. See error

**Technical information**
 - Device: honor9 lite
 - OS: android
 - OpenTracks version: 4.3.2
 - gadgetbridge version 0.73

dennisguse, 2023-05-03:
Do you have an error message or a logcat?

Dambond, 2023-05-05:
> Do you have an error message or a logcat?

Yep but when i try to past it it only paste : opentracks crashes (in my linguage)

Dambond, 2023-05-08:
> I meant a log file ;)
> 
> https://developer.android.com/tools/logcat

Is this ok?
### App information
* ID: de.dennisguse.opentracks
* Version: 5189 v4.3.2

### Device information
* Brand: HONOR
* Device: HWLLD-H
* Model: LLD-L21
* Id: HONORLLD-L21
* Product: LLD-L21

### Firmware
* SDK: 28
* Release: 9
* Incremental: 9.1.0.156C636

### Cause of error
```java
Exception in thread "main": java.lang.RuntimeException: Unable to start activity ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.publicapi.StopRecording}: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=de.dennisguse.opentracks/.services.TrackRecordingService }: app is in background uid UidRecord{8d3a352 u0a127 TPSL idle change:idle|cached procs:1 seq(0,0,0)}
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3430)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3614)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:86)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:108)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:68)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2199)
	at android.os.Handler.dispatchMessage(Handler.java:112)
	at android.os.Looper.loop(Looper.java:216)
	at android.app.ActivityThread.main(ActivityThread.java:7625)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)
Caused by: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=de.dennisguse.opentracks/.services.TrackRecordingService }: app is in background uid UidRecord{8d3a352 u0a127 TPSL idle change:idle|cached procs:1 seq(0,0,0)}
	at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1810)
	at android.app.ContextImpl.startService(ContextImpl.java:1765)
	at android.content.ContextWrapper.startService(ContextWrapper.java:664)
	at android.content.ContextWrapper.startService(ContextWrapper.java:664)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.startAndBind(TrackRecordingServiceConnection.java:83)
	at de.dennisguse.opentracks.publicapi.AbstractAPIActivity.onCreate(AbstractAPIActivity.java:41)
	at android.app.Activity.performCreate(Activity.java:7458)
	at android.app.Activity.performCreate(Activity.java:7448)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1286)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3409)
	... 11 more
```

dennisguse, 2023-05-08:
@Dambond okay - I am not sure I understand why it is failing.
Could you create a video recording of what is happening?

Dambond, 2023-05-09:
I Will try, I have to records the band on the same time. It crash when my band finish the recording.

dennisguse, 2023-05-16:
@Dambond any update?

closed 2023-05-20 by dennisguse