OpenTracksApp/OpenTracks #1554: Error when trying to change recording layout

nixeo, 2023-05-05:
Step to reproduce:
1. Click record
2. Click menu (3 dots) & choose 'select layout' 
3. Choose any custom layout

### App information
* ID: de.dennisguse.opentracks
* Version: 5194 v4.3.3

### Device information
* Brand: samsung
* Device: o1s
* Model: SM-G991B
* Id: TP1A.220624.014
* Product: o1sxxx

### Firmware
* SDK: 33
* Release: 13
* Incremental: G991BXXU7EWCH

### Cause of error
```java
Exception in thread "main": java.lang.IllegalArgumentException: columnCount must be greater than or equal to the maximum of all grid indices (and spans) defined in the LayoutParams of each child. 
	at androidx.gridlayout.widget.GridLayout.handleInvalidParams(GridLayout.java:811)
	at androidx.gridlayout.widget.GridLayout$Axis.setCount(GridLayout.java:1173)
	at androidx.gridlayout.widget.GridLayout.setColumnCount(GridLayout.java:423)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment.onLayoutChanged(StatisticsRecordingFragment.java:125)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment.lambda$new$0(StatisticsRecordingFragment.java:66)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment.$r8$lambda$tMysS2hXjodPYdIchQvJybasmtA(Unknown Source:0)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment$$ExternalSyntheticLambda0.onSharedPreferenceChanged(Unknown Source:2)
	at android.app.SharedPreferencesImpl$EditorImpl.notifyListeners(SharedPreferencesImpl.java:693)
	at android.app.SharedPreferencesImpl$EditorImpl.apply(SharedPreferencesImpl.java:559)
	at de.dennisguse.opentracks.settings.PreferencesUtils.setString(PreferencesUtils.java:174)
	at de.dennisguse.opentracks.settings.PreferencesUtils.setDefaultLayout(PreferencesUtils.java:801)
	at de.dennisguse.opentracks.TrackRecordingActivity.lambda$onOptionsItemSelected$5(TrackRecordingActivity.java:283)
	at de.dennisguse.opentracks.TrackRecordingActivity$$ExternalSyntheticLambda3.onClick(Unknown Source:2)
	at com.android.internal.app.AlertController$AlertParams$3.onItemClick(AlertController.java:1485)
	at android.widget.AdapterView.performItemClick(AdapterView.java:376)
	at android.widget.AbsListView.performItemClick(AbsListView.java:1287)
	at android.widget.AbsListView$PerformClick.run(AbsListView.java:3563)
	at android.widget.AbsListView.onTouchUp(AbsListView.java:4733)
	at android.widget.AbsListView.onTouchEvent(AbsListView.java:4448)
	at android.widget.ListView.onTouchEvent(ListView.java:1709)
	at android.view.View.dispatchTouchEvent(View.java:15544)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3310)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2974)
	at android.widget.AbsListView.dispatchTouchEvent(AbsListView.java:10733)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3316)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2990)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3316)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2990)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3316)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2990)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3316)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2990)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3316)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2990)
	at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:1117)
	at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1971)
	at android.app.Dialog.dispatchTouchEvent(Dialog.java:1332)
	at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:1075)
	at android.view.View.dispatchPointerEvent(View.java:15803)
	at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:8153)
	at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:7877)
	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:7213)
	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:7270)
	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:7236)
	at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:7434)
	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:7244)
	at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:7491)
	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:7217)
	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:7270)
	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:7236)
	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:7244)
	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:7217)
	at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:10788)
	at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:10676)
	at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:10632)
	at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:10926)
	at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:285)
	at android.os.MessageQueue.nativePollOnce(Native Method)
	at android.os.MessageQueue.next(MessageQueue.java:335)
	at android.os.Looper.loopOnce(Looper.java:186)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8757)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
```

dennisguse, 2023-05-05:
Can reproduce.

Happens when switching from layout with three columns to 2 columns.

dennisguse, 2023-05-09:
Fix didn't fix it.

dennisguse, 2023-05-09:
Actually it is fixed :)

nixeo, 2023-05-09:
im using the fdroid version... let me download the nightly version & test it. Hold on.

nixeo, 2023-05-09:
> Fix didn't fix it.

Tried on the nighlty-build without any issue on the layout. From layout 2 column to 3 & back to 2 column.

Everything looks good. Thank you!

dennisguse, 2023-05-10:
Thanks for testing :)

commit 427048ac (2023-05-08): Bugfix: crash when changing layout from one with 3 column layout to one with 2 columns.

Fixes #1554.
closed 2023-05-16 by dennisguse