OpenTracksApp/OpenTracks #1561: Capturing Images causes crashing the app

JulianOestreich90, 2023-05-13:
Trying to make pictures while recording results in this error and crashing the app.

### App information
* ID: de.dennisguse.opentracks
* Version: 5196 v4.3.4

### Device information
* Brand: google
* Device: oriole
* Model: Pixel 6
* Id: TQ2A.230505.002
* Product: oriole

### Firmware
* SDK: 33
* Release: 13
* Incremental: 23409000

### Cause of error
```java
Exception in thread "main": android.content.ActivityNotFoundException: No Activity found to handle Intent { act=android.media.action.IMAGE_CAPTURE flg=0x3 clip={text/uri-list hasLabel(0) {U(content)}} (has extras) }
	at android.app.Instrumentation.checkStartActivityResult(Instrumentation.java:2204)
	at android.app.Instrumentation.execStartActivity(Instrumentation.java:1843)
	at android.app.Activity.startActivityForResult(Activity.java:5506)
	at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:728)
	at androidx.core.app.ActivityCompat$Api16Impl.startActivityForResult(ActivityCompat.java:809)
	at androidx.core.app.ActivityCompat.startActivityForResult(ActivityCompat.java:246)
	at androidx.activity.ComponentActivity$2.onLaunch(ComponentActivity.java:243)
	at androidx.activity.result.ActivityResultRegistry$2.launch(ActivityResultRegistry.java:175)
	at androidx.activity.result.ActivityResultLauncher.launch(ActivityResultLauncher.java:47)
	at de.dennisguse.opentracks.ui.markers.MarkerEditActivity.createMarkerWithPicture(MarkerEditActivity.java:247)
	at de.dennisguse.opentracks.ui.markers.MarkerEditActivity.onOptionsItemSelected(MarkerEditActivity.java:199)
	at android.app.Activity.onMenuItemSelected(Activity.java:4442)
	at androidx.activity.ComponentActivity.onMenuItemSelected(ComponentActivity.java:529)
	at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:352)
	at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:269)
	at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:110)
	at androidx.appcompat.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:66)
	at androidx.appcompat.widget.Toolbar$1.onMenuItemClick(Toolbar.java:225)
	at androidx.appcompat.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:781)
	at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:836)
	at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:159)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:987)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:977)
	at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:625)
	at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:156)
	at android.view.View.performClick(View.java:7506)
	at android.view.View.performClickInternal(View.java:7483)
	at android.view.View.-$$Nest$mperformClickInternal(Unknown Source:0)
	at android.view.View$PerformClick.run(View.java:29357)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7884)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
```

dennisguse, 2023-05-13:
We can prevent the crash, but this will not enable this functionality for you.
The problem is that there is no (compatible) camera application installed.
Precisely this:
```
        Intent intent = new Intent(MediaStore.ACTION_IMAGE_CAPTURE)
                .putExtra(MediaStore.EXTRA_OUTPUT, photoUri);
```

Did uninstall the camera application?

JulianOestreich90, 2023-05-13:
I'm on CalyxOs and I'm using Google Camera. I also installed now Open Camera. Still the same error.

dennisguse, 2023-05-13:
I created a PR that prevents the crash, when you try to capture an image.

For the image capture, the following intent is used: https://developer.android.com/reference/android/provider/MediaStore#ACTION_IMAGE_CAPTURE

You may need to check with CalyxOS why this is not present

PS/ A workaround might be to select an image from the gallery.

PR #1562 (2023-05-13): Bugfix: prevent crash when creating a marker (no camera or gallery ap…
…plication present).

Fixes #1561.
commit 29568d0e (2023-05-13): Bugfix: prevent crash when creating a marker (no camera or gallery application present).

Fixes #1561.
closed 2023-05-13 by dennisguse