OpenTracksApp/OpenTracks #1563: App Shortcuts for OpenTracks

christophrrb, 2023-05-13:
**Is your feature request related to a problem? Please describe.**
No problem. Just that OpenTracks has no app shortcuts.

**Describe the solution you'd like**
When the app icon is pressed and held, app shortcuts to start recording and to view statistics would show.

**Describe alternatives you've considered**
Considered using Tasker but thought that an app shortcut made more sense.

**Additional context**
I've created the code for it [in a fork here](https://github.com/christophrrb/OpenTracks) (and it least is works in  debug; haven't tried it for the other release versions), but between 8 and 9 of the tests fail when I run them. Some only fail when all the tests are run (deleteTrack and editTrackRecording for example) while exportImport tests have failed every time.

dennisguse, 2023-05-13:
Can you open a PR for this?

And which tests are failing?

PS/ AndroidStudio beta is required due to AGP 8.1.0

dennisguse, 2023-05-13:
And do you also want to implement pause/stop?

christophrrb, 2023-05-14:
Just created a pull request here https://github.com/OpenTracksApp/OpenTracks/pull/1564, and sure, I'll try to implement pause/stop in about a few hours.

I'm downloading Android Studio Beta now to see if that helps me.

christophrrb, 2023-05-14:
Updated the pull request. I think I have some files that need to be deleted in it (like the build.gradle file).

I was able to get an emulator up in AS Beta, and through the emulator all the tests passed except `testBuildUniqueFileName_exist()`.

I added the stop functionality, but since I didn't see a "start/stop" string, it still just says "Record". If there is no recording, it starts one, and if one is ongoing, it ends it.

dennisguse, 2023-05-14:
Neat! If you need stop as a translation just add the key :)

christophrrb, 2023-05-14:
Thanks!

And sorry, what do you mean by "add the key"? I wanted to just combine the words "Record" and "Stop", but in XML, I don't know how to concatenate string resources (if that can even be done). :/

dennisguse, 2023-05-15:
Now I understood it. 
The shortcuts you added are static.
This will work nicely for the agregated statistics. 

For the record start/stop, you are using the Public API to trigger the action. 
This has two downsides:
* The Public  API is disabled by default: this might be misleading to a new user (and it is a security issue; not a big one)
* The shortcuts do not show the actual status (aka recording vs not recording) and this is likely misleading - especially if no notification is required anymore (Android 13)

For this functionality dynamic shortcuts would be the likely the better way to go: https://www.google.com/url?q=https://developer.android.com/develop/ui/views/launch/shortcuts/creating-shortcuts&sa=U&ved=2ahUKEwivpc-lkvj-AhWSxAIHHdXZCrsQFnoECAAQAw&usg=AOvVaw24ljgtErX8P2V8N_GH3bx-

Can I suggest that we add the aggregated shortcut via static shortcut (aka xml) and check how dynamic shortcuts can be used.
What do you think?