OpenTracksApp/OpenTracks #1570: Feature Request: Option to use GPS elevation instead of barometric sensor

groutr, 2023-05-24:
When tracking routes with OpenTracks, my elevation gain/loss are absurdly large.
There seem to be a couple of other issues that brought up this issue:
https://github.com/OpenTracksApp/OpenTracks/issues/1108
https://github.com/OpenTracksApp/OpenTracks/issues/1278
https://github.com/OpenTracksApp/OpenTracks/issues/1254

Gaining 20k ft of elevation in a mile is clearly incorrect behavior. One of the issues above mentioned in a comment that elevation data is taken from the barometric sensor. It would be nice to have the option to use GPS elevation instead of the barometric sensor to measure gain/loss.

dennisguse, 2023-05-24:
We had some trouble with "broken" sensors - see README_SENSORS.md

Elevation gain/loss is purely measured using the barometric pressure sensor. 
Does OpenTracks measure gain/loss if you shake your device?

dennisguse, 2023-05-24:
And measuring gain/loss via GPS is kind of complicated as the data is very noisy. 
Almost everybody is using a barometer and combining it with the GPS data. 

PS/ I am working on adding support for external barometric sensors (as my phone doesn't have any). Here we could add the option to not use an internal sensor at all.

groutr, 2023-05-24:
I should also explain why I created this issue. I took a walk around my neighborhood last night and OpenTracks is convinced that I climbed/descended the second highest peak on earth (a gain of 28,701ft of elevation gain and 28,248ft of loss in 1.36 miles).

dennisguse, 2023-05-24:
The PlayService provide sensor fusion for GPS elevation and barometric sensors.
So, applications using it may be able to counter quite some invalid measurements.
Sadly, the PlayService are not open-source, so they are not an option for OpenTracks.

Without a (functioning) barometer you have two options:
* try to filter the GPS signal; here is an investigation on how to do this https://www.gpsvisualizer.com/tutorials/elevation_gain.html
* or using the SRTM/DEM data to derive the altitude above ground from the GPS coordinates

AAT supports the 2nd approach if you want to try it: https://f-droid.org/packages/ch.bailu.aat/ 

For the OpenTracks implementation: you can take a look at the AltitudeSumManager (exponential filtering + 3m threshold).
Noticing fancy and it gets the job sufficiently done.

And can you take a look at the interval tab?
Is the gain/loss distributed equally or at the beginning?

groutr, 2023-05-24:
[Intervals are listed 3rd sentence.](https://github.com/OpenTracksApp/OpenTracks/issues/1570#issuecomment-1561715887)

Looking back at some historical tracks, it seems like this is the first track with obviously bad elevation measurements. Has elevation code changed since March 1, 2023? I used opentracks to track several days of skiing earlier this year and elevations on those intervals seem very plausible (because I know roughly the elevataion gain of each lift).

groutr, 2023-05-24:
I might have edited the above comment after you read it:

> You mentioned that GPS elevation data is very noisy, another approach might be to investigate Kalman filtering (https://en.wikipedia.org/wiki/Kalman_filter#Example_application)?

dennisguse, 2023-05-24:
IIRC the code didn't change since it was implemented (except refactoring).

For GPS-based gain/loss: this is the best article I have found: https://www.gpsvisualizer.com/tutorials/elevation_gain.html
It is worth reading.

closed 2023-07-26 by dennisguse