OpenTracksApp/OpenTracks #1573: The wording of GPS accuracy

macias, 2023-05-29:
*Thank you for very interesting app*

**Describe the bug**

The wording of GPS accuracy is odd to me.

**To Reproduce**
1. Go to application settings
2. Select "GPS" section
3. Look at the description of "GPS accuracy"
4. It says "Location is ignored if its accuracy is **less** than X m".

IMHO it should read "more", because let's say you have set limit as 10m. Now you acquire position with 9 m accuracy, it is pretty solid, yet it is rejected because it is too accurate? It should be rejected if position has 11 m accuracy (more than limit).

dennisguse, 2023-05-29:
Yes, you are right!
Nice finding ;)

Can you create a pull request?

macias, 2023-05-29:
As for the pull request, only blindly, I didn't proceed further because there is information (no surprise) about checking tests and all. So I would like to help, but in this very moment (I have to pack before my trip) I won't be able to configure all environment necessary.

I am sorry for that.

dennisguse, 2023-05-29:
@macias don't worry. 
It should just be a change in the strings.xml (without any tests to be changed).
Will take care about this.

Enjoy your vacation!

dennisguse, 2023-05-31:
Done aka fix en-US and removed all it's translations.

commit c702c5a4 (2023-05-31): Bugfix: translation for settings_recording_min_required_accuracy_summary needs to be inverted.

Fixes #1573.
closed 2023-05-31 by dennisguse