OpenTracksApp/OpenTracks #1580: Feature request: Idle speed threshold based on activity

lcarrabba, 2023-06-02:
Exactly as happened for units preferred rate (Settings -> Default units and Activity -> Preferred rate) it would be very useful to set GPS -> Idle speed threshold with different value, based on sport. 
For example, walking could be a good standard threshold of 1 km/h, instead biking could standard set to 4km/h. Obviously there still be the possibility to set a custom value.

dennisguse, 2023-06-04:
I guess that brings in the direction that we need real profiles per activity type, right?

LuCarhub, 2023-06-04:
Hi Dennis,

I think it could be a good solution. It's also what I saw on sports tracker (ex Nokia Sports Tracker). And maybe create less activities, or the possibility to remove unuseful activities. 
What do you think about it?

dennisguse, 2023-08-03:
Will be superseeded by #1187 as this will introduce idle detection based upon "time not moved' rather than "speed below threshold".

closed 2023-08-03 by dennisguse