OpenTracksApp/OpenTracks #1583: Make KML/KMZ export compatible with GoogleMaps

macias, 2023-06-04:
**Is your feature request related to a problem? Please describe.**

GM is not a standard as such, but considering it is used quite widely it would be good to ensure KML/KMZ export uses such features GM is capable importing. Let's say I have a track, just a line and a waypoint. It is not very complex, yet on one case GM refused to import it at all, on the other it imported only waypoints ignoring the line.

Using GPX works fine.

**Describe the solution you'd like**

Basic trace (line + waypoints) should be possible to export from OT and import to GM. 

**Describe alternatives you've considered**

Sticking to GPX.

**Additional context**

Example: https://drive.google.com/file/d/1YkBhDJFf0WQq4zw5kd766yIAnDjCDKdf/view?usp=sharing

dennisguse, 2023-06-04:
Do you have an error message?
And on which platform are you?

By the way OpenTracks export KML 2.3

macias, 2023-06-04:
For the attached files I don't get any error, GM simply import waypoints only. On previous data (not included here) I didn't get error from OT, but I did get error from GM (sadly GM uses very generic error message, so you don't know what is the problem really, only a text about malformed file).

I use Android 13, Samsung XCover 5, and my computer is Linux openSUSE Leap 15.3.

The version could be a reason, after brief googling I see a lot of references to 2.2, but not 2.3.

dennisguse, 2023-06-08:
Likely Google didn't bother supporting KML2.3.
KML2.3 was adding custom extensions Google created for KML2.2 and making them a proper standard.
For OpenTracks, it would be too much effort to also export KML2.2 - imports are supported though.

closed 2023-06-08 by dennisguse