OpenTracksApp/OpenTracks #1584: incorrect distance announced with TTS

rvandegrift, 2023-06-04:
**Describe the bug**
During a run, I noticed that OpenTracks announces incorrect distance sometimes.  It announced "1 mile" twice, once at 1.46 and again at 1.97.  The data and screen display seemed correct.  Other announcements included the decimal places, so it isn't consistent.

**To Reproduce**
1. Enable announcements that include Total Distance
2. Go for a run and listen

**Technical information**
 - Device: Pixel 5a
 - OS: LineageOS 20
 - OpenTracks version: v.4.3.5

dennisguse, 2023-06-04:
What is your announcement interval?

Sounds like it is rounded down to 1 mile.

rvandegrift, 2023-06-05:
The announcement interval is 5 minutes.  Could be rounding, but it's definitely not always.  So far I haven't identified a pattern - hard to troubleshoot during a run :)

dennisguse, 2023-06-05:
As you using the time interval: just record a track with 1.5 miles and stop it.

Then when you are indoors, just continue it.
As you won't move, OpenTracks will just announce at the time interval (which you can also reduce).

Hypotheses:
* OpenTracks has a problem with miles (a rather exotic measurement unit)
* OpenTracks has a problem with your locale (which language are you using?)
* something is broken in TTS of your phone (rather unlikely)
* or something else :D

dennisguse, 2023-06-11:
@rvandegrift any news?

rvandegrift, 2023-06-11:
@dennisguse sorry for the slow response!  For tracks >=2 miles, OpenTracks consistently rounds to the nearest 1/10 of a mile.  It appears to truncate distances < 1mi.

I tried continuing a few tracks with 1 min announcement intervals, here's what I heard:
- 1.25 miles announced "1 mile"
- 1.65 miles announced "1 mile"
- 2.26 miles announced "2.3 miles"
- 2.98 miles announced "3.0 miles"
- 4.70 miles announced "4.7 miles"
- 15.40 miles announced "15.4 miles"

I believe I've heard announcements in the 1<x<2 range with a decimal.  But I can't reproduce that now (I only found two tracks short enough, plus I've f-droid upgraded to 4.4.0 since I opened this.)

All of this is using English - US English, as you can image from my usage of this exotic unit!

dennisguse, 2023-06-11:
That's something to work with. 
Let me see if I can reproduce this :)

dennisguse, 2023-06-12:
I can reproduce this for ALL voice announcements having `1.X`; all of them are rounded down to `1 unit`.
The total distance the problem is here: 
`
            appendDecimalUnit(builder, context.getResources().getQuantityString(distanceId, getQuantityCount(distanceInUnit), distanceInUnit), distanceInUnit, 1, unitDistanceTTS);
`

where `getQuantityCount()` returns `1` (always rounding down) and then `getQuantityString()` selects the appropriate plural.
BUT our plurals look like this:
```
    <plurals name="voiceDistanceMiles">
        <item quantity="one">1 mile</item>
        <item quantity="other">%1$.1f miles</item>
    </plurals>
```
For 1, we have no variable in there...

Not sure how to fix this properly.
Question is `1.1 mile` or `1.1 miles` aka is quantity=1 or quantity=2?
And we might have the problem that this is different in various languages.

dennisguse, 2023-06-12:
Yeah: https://stackoverflow.com/questions/54882981/android-plurals-for-float-values](https://stackoverflow.com/questions/54882981/android-plurals-for-float-values)

Moreover, our translations are very inconsistent; some have a hardcoded one, some have a variable...
That adds the weirdness of this issue.
@rvandegrift Nice finding by the way :sunglasses:

rvandegrift, 2023-06-12:
heh - thanks :)

> Question is 1.1 mile or 1.1 miles aka is quantity=1 or quantity=2?

To me, all miles with decimals should be plural: 0.2 miles, 1.1 miles, etc.  Even "1.0 miles".  I think singular sounds right only when it's exactly "1 mile".

That looks like just using the `other` clause you quoted above?

dennisguse, 2023-06-13:
I found the solution in the Android's docs: https://developer.android.com/guide/topics/resources/string-resource

[image]

Here is a proof of concept (also failing tests for the current implementation: https://github.com/OpenTracksApp/OpenTracks/tree/proofOfConcept%231584

**Solution:** use ICU MessageFormat (not Android's `getQuantityString()`).
For this, we have to 
1. increase minSDK Version from 21 to 24 (aka Android 7.0).
2. convert all plurals to this new format for all `strings.xml`

Some more input: https://phrase.com/blog/posts/android-app-development-tutorial-multi-language-support/

@pstorch @rgmf Any input on increasing minSDK to 24?
It is the same as OSMDashboard.
Alternative solution: keep if's to only use `getCountyString()` below 24 and duplicate the plurals (<24 and >=24).

rgmf, 2023-06-13:
@pstorch @rgmf Any input on increasing minSDK to 24?
It is the same as OSMDashboard

It makes sense. I'd do it.

commit 7c292373 (2023-06-21): ICU's MessageFormat handles plurals for floats (getQuantityString does not).

Fixes #1584
commit 1f86250e (2023-06-21): ICU's MessageFormat handles plurals for floats (getQuantityString does not).

Fixes #1584.
closed 2023-06-21 by dennisguse