OpenTracksApp/OpenTracks #1585: Make tagging easier

macias, 2023-06-05:
**Is your feature request related to a problem? Please describe.**

While UI is intuitive the problem is adding markers is younger brother compared to live statistics. I use OSM Tracker a lot, and there adding marker is single tap/click ("add marker") on the big button.

In OT it takes more time and requires more precision -- it is three clicks (menu, add marker, activate keyboard for typing) and the area is smaller. The size is important IMHO as well because in full sun I often tap a bit off, so on few occasions I clicked on the wrong item.

**Describe the solution you'd like**

I don't know how such change is aligned with mission "sport application", but the first approach could be to add intermediate step :-).

Currently: main screen with live stats -> menu button -> "add marker" button.

Suggested approach: main screen with live stats -> menu button -> "tagging view" -> add marker.

Such change would allow user to switch view once per day (or life) and stick to the tagging for long. Then adding marker would be just one click/tap.

Thinking aloud -- among next steps I would see creating marker buttons, like "asphalt", "gravel", "grade4", etc. by user. They would appear (once defined) on "tagging view" and it would be just one tap to add such marker. Other thing would be triggering keyboard right after "add marker" is tapped/clicked. List of added tags would be great with clearly shown the last one, so for example if I mis-clicked "asphalt" I could spot my mistake, remove this marker, and click on "gravel" instead (just an example).

**Describe alternatives you've considered**

Instead of going through menu -> "tagging view" it could be also handy to have blinds-buttons (collapsible views) or tabs. But both take space, so on big screen it would be better, but not on a phone (IMHO).

dennisguse, 2023-06-05:
Something like this?

https://m3.material.io/components/floating-action-button/overview

dennisguse, 2023-06-07:
I meant the floating action button

macias, 2023-06-07:
It could be done this way, but I don't see how it is improvement. I.e. I don't see a big picture here -- would FAB be used for switching view? Then it would be annoyance. If FAB button would be used to add marker on "tagging view". Ok, but then it does not scale -- because once there would be customizable tags, you would need more FABs, and as I understand this type of button works well, when you have scrollable view + one button, like list of contacts with "(+)" button for adding new one.