OpenTracksApp/OpenTracks #1588: When importing gpx workouts, the statistics are missing (important for the athlete) heart rate (min.max.average)

snooppr, 2023-06-06:
There is a graph for the pulse, but it's meaningless (no % in zones, no connecting, minimum, maximum values)

In general, for an athlete, there is not enough in the statistics: number of steps;  min.max.  total climb;  pulse;  calories burned.  
This lack defines a sports track from a simple tracker.

closed 2023-06-06 by dennisguse

snooppr, 2023-06-06:
About height.  The same track in open source, fito track calculation shows no in OT.

[image]

snooppr, 2023-06-06:
@dennisguse HI, 
> If you have time, both things just wait for somebody to implement them.

Perhaps the question will seem strange, but why does the developer hope that someone else implements the main functionality for OT, but not himself?  (OT is advertised as a sports tracker, which is confusing when migrating).  From the point of view of open source, this is almost the norm, but from the point of view of the user, it is not the norm.

dennisguse, 2023-06-06:
Free as in speech not as in beer.

I build things in my spare time and if this accidentally is useful to somebody else, this is a bonus. 
If somebody proposes something that I like to build (because personal reasons), I might build it. 
If not then not.

Why others  contribute (code or translations or whatever) is their concern. 

If you really want something done, you either need to do it yourself or motivate others to do it for you. 

PS I would be available as a (paid) freelancer