OpenTracksApp/OpenTracks #1599: Crash on start through intent

sgofferj, 2023-06-11:
**Describe the bug**
Crash on start through intent

**To Reproduce**
1. Send a start intent

### App information
* ID: de.dennisguse.opentracks.nightly
* Version: 5249 v4.4.0-23-g0b8c2a0ef

### Device information
* Brand: ulefone
* Device: Armor_3
* Model: Armor_3
* Id: O11019
* Product: Armor_3

### Firmware
* SDK: 27
* Release: 8.1.0
* Incremental: 1535356309

### Cause of error
```java
Exception in thread "main": java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.sensors.AltitudeSumManager.fill(de.dennisguse.opentracks.data.models.TrackPoint)' on a null object reference
	at de.dennisguse.opentracks.sensors.SensorManager.fill(SensorManager.java:81)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.addSensorData(TrackPointCreator.java:68)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.onNewTrackPoint(TrackPointCreator.java:89)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.onChange(TrackPointCreator.java:84)
	at de.dennisguse.opentracks.sensors.SensorManager$1.onChange(SensorManager.java:30)
	at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.onChanged(BluetoothRemoteSensorManager.java:168)
	at de.dennisguse.opentracks.sensors.AbstractBluetoothConnectionManager.connect(AbstractBluetoothConnectionManager.java:159)
	at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.connect(BluetoothRemoteSensorManager.java:160)
	at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.-$$Nest$mconnect(Unknown Source:0)
	at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager$1.onSharedPreferenceChanged(BluetoothRemoteSensorManager.java:75)
	at de.dennisguse.opentracks.settings.PreferencesUtils.registerOnSharedPreferenceChangeListener(PreferencesUtils.java:86)
	at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.start(BluetoothRemoteSensorManager.java:116)
	at de.dennisguse.opentracks.sensors.SensorManager.start(SensorManager.java:55)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.start(TrackPointCreator.java:51)
	at de.dennisguse.opentracks.services.TrackRecordingService.startSensors(TrackRecordingService.java:207)
	at de.dennisguse.opentracks.services.TrackRecordingService.startRecording(TrackRecordingService.java:194)
	at de.dennisguse.opentracks.services.TrackRecordingService.startNewTrack(TrackRecordingService.java:175)
	at de.dennisguse.opentracks.publicapi.StartRecording.execute(StartRecording.java:27)
	at de.dennisguse.opentracks.publicapi.AbstractAPIActivity.lambda$new$0(AbstractAPIActivity.java:22)
	at de.dennisguse.opentracks.publicapi.AbstractAPIActivity.$r8$lambda$iA-X-JpoEEQl7oKhXQKZcU1FoqI(Unknown Source:0)
	at de.dennisguse.opentracks.publicapi.AbstractAPIActivity$$ExternalSyntheticLambda1.onConnected(Unknown Source:2)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.setTrackRecordingService(TrackRecordingServiceConnection.java:147)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.onServiceConnected(TrackRecordingServiceConnection.java:162)
	at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:1662)
	at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:1691)
	at android.os.Handler.handleCallback(Handler.java:790)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6548)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:857)
```

**Technical information**
See crash log

dennisguse, 2023-06-11:
Fixed by: https://github.com/OpenTracksApp/OpenTracks/commit/0122539d0fecc9b599ba99cfd31198c8907f4283

PS/ thanks for using the nightly and reporting such things :+1:

closed 2023-06-11 by dennisguse

sgofferj, 2023-06-12:
> PS/ thanks for using the nightly and reporting such things +1

Well, my Java/Kotlin sucks, so that's the only way I can help.
That bug report button doesn't do anything, btw. It opens Github on the mobile browser but then Github says "something went wrong"

dennisguse, 2023-06-12:
Interesting that the link doesn't work.
Might be that the body became too long.
Can you get the URL the next time?

sgofferj, 2023-06-13:
Sure