OpenTracksApp/OpenTracks #1600: support for ble devices that don't pair.

nutpantz, 2023-06-11:
It seems there are a number of ble devices that will never pair to s smart phone, but they can still be used. 
(https://www.amazon.ca/POSMA-Bluetooth-Cycling-Computer-Monitor/dp/B07VSLL5HQ)

The app looks for and finds the device  connects and it just works.
Eg: 
https://github.com/jonasoreland/runnerup
(Setting-sensors-heart rate-experimental-Bluetooth le then scan)

dennisguse, 2023-06-11:
Actually, OpenTracks does not require BLE devices to be paired.
It just stores the address and when a recording is started, it tries to connect.

Or what is the suggestion?

nutpantz, 2023-06-11:
at the moment, open tracks has no method to look for devices that are not paired that i can see. 
If the device is not paired it never shows up on the list to select.. 
So there would be a scan for all and select to use option for heart rate (and any other devices) needed.

Ie I can be connect and watch  the info with nrf and nothing is listed in open tracks.

dennisguse, 2023-06-11:
Actually that is what OpenTracks is doing.
It scans for all devices that announce a certain BLE service (e.g., heartrate) and then it will be shown.
If the device doesn't announce the required service in it's announcements packets (like PineTime for heartrate), this filter can even be deactivated.
... but the list may get very long depending on where you are.

The filter can be deactivated in the settings page for BLE; it is the last one.

nutpantz, 2023-06-11:
Open tracks crashed and dumped a log
[image]

nutpantz, 2023-06-11:
This one can be closed, since it was an inaccurate issue

closed 2023-06-11 by nutpantz