OpenTracksApp/OpenTracks #1601: crash ble connect

nutpantz, 2023-06-11:
### App information
* ID: de.dennisguse.opentracks.nightly
* Version: 1 

### Device information
* Brand: Blackview
* Device: BV9800Pro
* Model: BV9800Pro
* Id: PPR1.180610.011
* Product: BV9800Pro_EEA

### Firmware
* SDK: 28
* Release: 9
* Incremental: root.20200814.112030

### Cause of error
```java
Exception in thread "main": java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.sensors.AltitudeSumManager.fill(de.dennisguse.opentracks.data.models.TrackPoint)' on a null object reference
	at de.dennisguse.opentracks.sensors.SensorManager.fill(SensorManager.java:81)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.addSensorData(TrackPointCreator.java:68)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.onNewTrackPoint(TrackPointCreator.java:89)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.onChange(TrackPointCreator.java:84)
	at de.dennisguse.opentracks.sensors.SensorManager$1.onChange(SensorManager.java:30)
	at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.onChanged(BluetoothRemoteSensorManager.java:168)
	at de.dennisguse.opentracks.sensors.AbstractBluetoothConnectionManager.connect(AbstractBluetoothConnectionManager.java:159)
	at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.connect(BluetoothRemoteSensorManager.java:160)
	at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.-$$Nest$mconnect(Unknown Source:0)
	at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager$1.onSharedPreferenceChanged(BluetoothRemoteSensorManager.java:75)
	at de.dennisguse.opentracks.settings.PreferencesUtils.registerOnSharedPreferenceChangeListener(PreferencesUtils.java:86)
	at de.dennisguse.opentracks.sensors.BluetoothRemoteSensorManager.start(BluetoothRemoteSensorManager.java:116)
	at de.dennisguse.opentracks.sensors.SensorManager.start(SensorManager.java:55)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.start(TrackPointCreator.java:51)
	at de.dennisguse.opentracks.services.TrackRecordingService.startSensors(TrackRecordingService.java:207)
	at de.dennisguse.opentracks.services.TrackRecordingService.tryStartSensors(TrackRecordingService.java:202)
	at de.dennisguse.opentracks.TrackListActivity.lambda$new$2(TrackListActivity.java:155)
	at de.dennisguse.opentracks.TrackListActivity.$r8$lambda$i5o90iCDb0De3uoVzrIHAGyUQRc(Unknown Source:0)
	at de.dennisguse.opentracks.TrackListActivity$$ExternalSyntheticLambda10.onConnected(Unknown Source:2)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.setTrackRecordingService(TrackRecordingServiceConnection.java:147)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.onServiceConnected(TrackRecordingServiceConnection.java:162)
	at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:1730)
	at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:1762)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6702)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)
```

nutpantz, 2023-06-11:
I think this can be closed

nutpantz, 2023-06-11:
Dang t not this one the other one can be closed

nutpantz, 2023-06-11:
Don't have the error on 3.15 ps

nutpantz, 2023-06-11:
Ok it was just the custom nightly that was crashingb so this can be ignored. (I had no idea it was so custom)

closed 2023-06-11 by nutpantz

dennisguse, 2023-06-11:
See #1599.