OpenTracksApp/OpenTracks #1603: total over time average heart rate since start display.

nutpantz, 2023-06-11:
As of right now it appears that open tracks only shows the present heartrate. (If the setting is there i can't find it so far) Average heart rate since start is very important for gauging over all fitness for those of us who don't want to get crazy complicated.

It would give someone at a glace an idea if their body thinks the workout is less effort at from times or locations with just one number.(average heartrate)

dennisguse, 2023-06-11:
I guess #185 might be better.

closed 2023-06-11 by dennisguse

nutpantz, 2023-06-11:
Well i would love to have a kitchen sink added also,  for myself, just to record and give an average for the workout word be enough.
Something like record the heart rate every 15 secs and give the average of all up to now. And the average at the end.

Show it just like heart rate just put "avg" in front of the new one

That would be good enough for me.

Now sure all the other options would be nice, but if the heart rate is recorded with every time entry(with or without gps). There must be some software someplace that can do the details.
(Or save info as a trx Garmin workout format as an option), there are lots of desktop apps that can play with that

If there is more demand later that is later