OpenTracksApp/OpenTracks #1611: moving time is incorrect.

nutpantz, 2023-06-18:
**Describe the bug**
The moving time vs total time is always incorrect.
95% of my tracks are 100% moving. But there is always a big difference between moving time and total time, sometimes very significant.
Possibly caused by trees ( not sure how it could be this bad since the gps is very very sensitive)

dennisguse, 2023-06-22:
What where you doing (aka which speed)?
And how was the landscape (aka GPS reception)?

nutpantz, 2023-06-22:
Speed was about 5-6 kph average 90% of the time. 30% tree coverage that never seems to reduce gps signal when i check it (under 3 meter fix) the rest under open sky (1 meter fix)

dennisguse, 2023-06-22:
Problem: OT uses (GPS) speed to determine moving/non-moving only.
(GPS) speed is very unreliably - especially when being slow aka walking.

Would be fixed by #1187.

closed 2023-06-22 by dennisguse