OpenTracksApp/OpenTracks #1619: global waypoints for all tracks and maps

nutpantz, 2023-06-26:
I use waypoints a lot, i track trail heads parking lots, gates, bridges streams, bathrooms, etc. However the waypoints in open tracks are virtually unused because they are only with each track.
Having the ability to have global waypoints wood be fantastic. Then without leaving the app i could see previous important features.
A simple quick waypoint system similar to some other apps of ask that is needed
[image]

Where you could quickly add a waypoint that would be visible on all maps during all tracks. 
Maybe even a simple directory for editing
[image]

dennisguse, 2023-06-26:
The 2nd screen, we have.
It is available from the track list.
One issue here is that OpenTracks requires that each marker/waypoint belongs to a track.
And that might be tricky to change.

nutpantz, 2023-06-26:
Maybe build a global waypoint system into the osm dash board? Extend it into a standalone generic map waypoint and track viewer for any gpx, kml etc. 
Then open tracks would not become to bloated with features but the dashboard wood have much more value as a standalone app.

Just spitballing ideas

pstorch, 2023-07-01:
Should we put this idea into the https://github.com/OpenTracksApp/OSMDashboard repo?
I don't know if it fits there. At the moment the OSMDashboard doesn't have a database. And I don't know if both apps should have storage for waypoints.

nutpantz, 2023-07-01:
Well it's really about where the author wants his apps to go. If he is interested in osmdashboard becoming more then just a map for open tracks.  Maybe even a map and waypoint manager for any number of gps apps. 
The functions could also be rolled into OpenTracks also as part of the fitness tracking, since i think global waypoints and not track only waypoints is a good idea.

So it's really what direction the author is interested in. 
There even could be an argument for the functions in both apps in a symbiosis type situation.