OpenTracksApp/OpenTracks #1620: Cannot add existing picture to marker

khao-soi, 2023-06-27:
**Describe the bug**
When I use the regular android camera app to take a picture of something and try to add that picture to a marker, it says "cancelled, no photo marker added". I also tried to add the picture via the internal photo function and the galery preview of the camera view, but I can only see the picture there but can't apply it as the markers image.

Or am I doing something wrong? What is a photo marker and how does it differ to a regular marker?

**To Reproduce**
1. Take a picture of anything
2. Have a track with a marker
3. Go to that marker
4. Click edit icon (pen)
5. Click add image icon (not take photo icon)
6. See error "cancelled, no photo marker added"

If applicable:
[image]

**Technical information**
 - Device: OnePlus 8T
 - OS: LineageOS 20.0
 - OpenTracks version: 4.5.0 (F-Droid)

dennisguse, 2023-06-27:
Do you press the button in the camera that you want the picture that was just taken?
What seems to happen is that the camera app reports that no picture was taken and therefore none will be transferred to OpenTracks.

Can you create a video of what is happening?

khao-soi, 2023-06-27:
I pressed the preview button in the camera with the picture just taken, yes. But that was just my attempt to workaround the not functioning "add image" button.

Here's a video:

https://github.com/OpenTracksApp/OpenTracks/assets/2349590/3e126554-884e-4a13-a0f6-6dbb0a64e932

dennisguse, 2023-06-27:
Mhh okay.
Not sure what happens; actually an gallery activity should be opening.
I mean the first case should be working and the 2nd cannot (only allows taking new pictures).

Relevant code is in `MarkerEditActivity`; look for all three cases of `R.string.marker_add_photo_canceled`.

Any chance to get a logcat?

dennisguse, 2023-06-27:
Or might be related to Android 13s permissions. 
https://github.com/Drjacky/ImagePicker/issues/73

khao-soi, 2023-06-27:
Here come logcat. One filtered by PID and one unfiltered.

[logcat_opentracks.txt](https://github.com/OpenTracksApp/OpenTracks/files/11885524/logcat_opentracks.txt)
[logcat.txt](https://github.com/OpenTracksApp/OpenTracks/files/11885526/logcat.txt)

khao-soi, 2023-06-27:
> @khao-soi can you try: #1621

Yeah I could. Give me some time to set up OpenTracks in my local dev env.

khao-soi, 2023-06-27:
> @khao-soi can you try: #1621

Android Studio doens't want to start it.

`The project is using an incompatible version (AGP 8.1.0-rc01) of the Android Gradle plugin. Latest supported version is AGP 8.0.2`

closed 2023-06-27 by dennisguse

dennisguse, 2023-06-28:
@khao-soi We currently use the RC of AndroidStudio (due to a nice feature in AGP 8.1.0).
Hope the stable release comes soon :D

PR #1621 (2023-06-27): Bugfix: Android 13 couldn't open Gallery.
Fixes #1620.

We should use the photo picker (especially from Android 13 onwards).
https://developer.android.com/training/data-storage/shared/photopicker
commit 336b9eaa (2023-06-27): Bugfix: Android 13 couldn't open Gallery.
Fixes #1620.