OpenTracksApp/OpenTracks #1623: Aggregated Stats for multiple selected tracks disappeared in v4.5.0

BosqueRider, 2023-06-28:
In previous versions of OpenTracks, including v4.4.0, when multiple tracks were selected,  the dropdown menu provided 3 options:  "Aggregated stats", "Delete", and "Select All".  With the latest update to v4.5.0, the "Aggregated stats" option is gone.

I used that function daily by highlighting all tracks for the day, and then selecting "Aggregated stats" to view all information for the day on a single screen. It was very helpful and convenient!

Was this an unintentional omission or is it necessary to revert to v4.4.0 to regain this functionality?

Thanks for providing a great privacy-respecting app!

Regards, BosqueRider

P.S.   If helpful, I can revert to v4.4.0 and provide screenshots.

dennisguse, 2023-06-29:
The functionality moved to the bottom bar. 

[image]

BosqueRider, 2023-06-29:
Thanks for the quick reply.  I'm aware of that button to view all aggregated stats, but that's not what I'm referring to.

 For instance, from the screen that shows all individual events, where I currently have 400+ trips recorded, I press and hold on an event to go into selection mode. Then, I can press other trips to highlight them.  For example, if I select 5 out of 400 events, previously the menu provided an option to show the aggregated stats for those 5 trips only.

 A few days ago, after upgrading via F-Droid to v4.5.0, that option was gone.

BosqueRider, 2023-06-29:
The menu shown in screen shot ...56-34 is where a 3rd option used to be included for aggregated stats.

[image]
[image]

BosqueRider, 2023-06-29:
In the example above, if I use the map button, only the two selected routes will be shown. Previously, the aggregated stats option from this screen would show the combined stats for those two segments only - a very handy feature!

dennisguse, 2023-06-29:
Re-added it :D

Will be in the next release (after v4.6.0).

closed 2023-06-29 by dennisguse

BosqueRider, 2023-06-29:
Thank you! LOVE this app!  For those who might wonder what I'm talking about, here's two more screenshots, along with Greetings from the Land of Enchantment, New Mexico, USA.
[image]
[image]

BosqueRider, 2023-09-29:
Hi Dennis,

THANK YOU (!!!) for the quick fix on this back in June. OpenTracks is a great app! I've used everyday for 347 consecutive days, with over 1100 trips recorded. It's THE BEST!!! There are a few tricks I've learned along the way to keep it consistently tracking. If you would like more details or if there's a good place to share those tips, please let me know.

Now for some fun!

I'm in Albuquerque, New Mexico, near the Rio Grande River, and in 8 days, the Albuquerque International Balloon Fiesta begins it's 51st year, with over 500 Hot Air Balloons registered and 12 new Special Shapes. This is a family-friendly must see event! It runs for 9 days, from October 7-15 (always from the first full weekend in October to through the second weekend), with sessions everyday as long as the weather cooperates. Everyday has a morning event, usually from 0600 to 0900, and evening events everyday except for Mon/Tue/Wed and the last Sunday, for a total of 14 sessions.

On the second Saturday, October 14, the Fiesta coincides with an Annular Solar Eclipse and a special daytime "Glow" with 72 balloons is planned. If the weather is good, which it usually is in this high desert climate, (Albuquerque's elevation is over 5,000 feet / 1,500 meters) it will be spectacular!

Why all this info? Because all of the Balloon Fiesta events can be streamed free, live online worldwide and I thought you might enjoy it with friends and family. If I remember correctly, you're in Berlin. This balloon, "KREFELD650", flown by Max Michels is from Germany. There's an 8-hour time difference between Berlin and Albuquerque - we're 8 hours earlier. Right now it's about 0200 here, and 1000 in Germany.

The main website:
https://balloonfiesta.com

Direct link to "Balloon Fiesta Live!"
https://balloonfiesta.com/Live-Stream

On a side note, I'm volunteering at the "Bike Valet." (1,200 volunteers "Navigators" are a crucial part of making this annual non-profit event happen.) Bicycle riders, both traditional pedal-bikes and e-bikes, can avoid the car traffic jams and the $20 parking fee and park for free right next to the field at the Bike Valet. I'm from Albuquerque, but hadn't tried this until last year. I should have done it earlier -- It's the very best way to get there!
https://balloonfiesta.com/Bike-Valet

Enjoy!

- Terry
- BosqueRider on github, as in the Rio Grande Bosque (the wooded area around the river), a fun place to ride!

Sent with [Proton Mail](https://proton.me/) secure email.

commit 7d7ab577 (2023-06-29): Track list: re-added aggregated stats for selections.

Removed in 9f1ea4d7450e417dfa7c995ce65e1d9c3402dcc2

Fixes #1623.