OpenTracksApp/OpenTracks #1625: Pure black/OLED theme missing in v4.5.0

ae5960e8-a6fc-491f-b252-898ecf59af95, 2023-06-30:
**Describe the bug**
The pure black/OLED theme isn't there anymore in the latest update.

**To Reproduce**
1. Go to User Interface > UI Theme
2. The black theme is missing

**Technical information**
 - OpenTracks version: v4.5.0

dennisguse, 2023-07-01:
There was no specific OLED theme -it was just the dark theme. 
Is this not working for you?

ae5960e8-a6fc-491f-b252-898ecf59af95, 2023-07-01:
I'm only seeing a dark purple theme as the dark theme, whereas I think there was a pure black toggle before.

dennisguse, 2023-07-01:
That might be the dynamic color feature of MaterialDesign. 
So, we may need to implement a separate oled friendly theme then.

dennisguse, 2023-08-05:
That's how it is done in F-Droid:
```
    /**
     * Apply pure black background in dark theme setting. Must be called in every activity's
     * {@link AppCompatActivity#onCreate()}, before super.onCreate().
     *
     * @param activity The activity to apply the setting.
     */
    public void applyPureBlackBackgroundInDarkTheme(AppCompatActivity activity) {
        final boolean isPureBlack = Preferences.get().isPureBlack();
        if (isPureBlack) {
            activity.setTheme(R.style.Theme_App_Black);
        }
    }
```

dennisguse, 2023-08-15:
@ae5960e8-a6fc-491f-b252-898ecf59af95 If you have time, just do it. 
It is not on my personal agenda (also due to summer).

commit e85045e1 (2024-02-13): UI: re-add OLED friendly (aka black) theme.

Fixes #1625.
commit 2c1af6d7 (2024-02-17): UI: re-add OLED friendly (aka black) theme.

Fixes #1625.
commit 04915e32 (2024-02-22): UI: re-add OLED friendly (aka black) theme.

Fixes #1625.
commit 9e59b0ee (2024-03-02): UI: re-add OLED friendly (aka black) theme.

Fixes #1625.
closed 2024-03-02 by dennisguse