OpenTracksApp/OpenTracks #1643: Overestimation of gain/loss

dennisguse, 2023-07-16:
The gain/loss seems to be inflated. 
Seemingly introduced in v4.5.0

I got a report by email.

DasFranck, 2023-07-22:
Hello,
I am also encountering this issue on a device with a barometer with a magnitude of around x10 the correct gain/loss.

Kind Regard,

dennisguse, 2023-07-26:
I cannot reproduce this problem :/
My good old Nexus 4 works still as expected (tested with v4.6.2).

DasFranck, 2023-07-26:
I currently have a rooted Pixel 7 with Android 13 running OpenTracks v4.6.1 (latest version available on F-Droid).

I will attach screenshots of OpenTracks' track summary highlighting the issue in a few hours.

DasFranck, 2023-07-26:
[image]
[image]

dennisguse, 2023-07-28:
@DasFranck that looks more like the barometric sensor has a problem. 
Did it work before (aka in an earlier version of OpenTracks)?
What happens if you shake the device?

DasFranck, 2023-07-28:
@dennisguse Barometric sensor seems fine, I have this device since only two weeks coming from a device that didn't had it.
I'm not sure to understand in which context I should shake my device.

However interestingly, the GPX files exported from those tracks are not impacted by this bug so if the barometric information are used for altitude in the GPX file this means that it's only a display bug on the track stats interface.

dennisguse, 2023-07-30:
@DasFranck I just wrote the procedure to check if the barometric sensor works for OpenTracks.
Can you check this: https://github.com/OpenTracksApp/OpenTracks/blob/main/README_TESTED_SENSORS.md#tested-barometric-sensor-smartphones

Altitude values from GPS; gain/loss from barometric sensor.

closed 2023-08-02 by dennisguse

DasFranck, 2023-08-02:
Barometer seems to go crazy only while rotating the phone upside down, normal in the other cases. Which can happen a lot while walking or biking and having the phone in a pants pocket.
I guess this means that the barometer inside the Pixel 7 are poorly calibrated or have a dubious precision.

Either way thanks for your time, I think there's nothing else to do on the app's side.

dennisguse, 2023-08-02:
To defend the Pixel 7: integrating a barometer is really tricky (waterproof, user touching the device, movement, protective cases)and it is -compared to other features- often neglectable. Most phones are not bike computers. My phone doesn't even have one. 
For this reason, I have the plan to add support for BLE barometers to OpenTracks.