OpenTracksApp/OpenTracks #1647: Crash when trying to share a workout

RugeljGG, 2023-07-28:
I'm not able to export (to email/file/Dropbox) the latest workout using "Share" button, as the app crashes with the following trace. This might have something to do with heart rate data, as I can export other workouts that were done without HR sensor.

### App information
* ID: de.dennisguse.opentracks
* Version: 5289 v4.6.1

### Device information
* Brand: samsung
* Device: r0s
* Model: SM-S901B
* Id: TP1A.220624.014
* Product: r0sxeea

### Firmware
* SDK: 33
* Release: 13
* Incremental: S901BXXS6CWF6

### Cause of error
```java
Exception in thread "AsyncTask #2": java.lang.RuntimeException: An error occurred while executing doInBackground()
	at android.os.AsyncTask$4.done(AsyncTask.java:415)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:381)
	at java.util.concurrent.FutureTask.setException(FutureTask.java:250)
	at java.util.concurrent.FutureTask.run(FutureTask.java:269)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
	at java.lang.Thread.run(Thread.java:1012)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference
	at de.dennisguse.opentracks.data.models.TrackPoint.getLatitude(TrackPoint.java:162)
	at de.dennisguse.opentracks.io.file.exporter.GPXTrackExporter.writeTrackPoint(GPXTrackExporter.java:311)
	at de.dennisguse.opentracks.io.file.exporter.GPXTrackExporter.writeTrackPoints(GPXTrackExporter.java:154)
	at de.dennisguse.opentracks.io.file.exporter.GPXTrackExporter.writeTrack(GPXTrackExporter.java:111)
	at de.dennisguse.opentracks.data.ShareContentProvider.lambda$openFile$2(ShareContentProvider.java:216)
	at de.dennisguse.opentracks.data.ShareContentProvider$$ExternalSyntheticLambda4.writeDataToPipe(Unknown Source:11)
	at android.content.ContentProvider$1.doInBackground(ContentProvider.java:2438)
	at android.os.AsyncTask$3.call(AsyncTask.java:394)
	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
	... 3 more
```

dennisguse, 2023-07-29:
Mhhh.

It is  not related to HR data (at least not directly).
There is a TrackPoint that should have a location (for GPX that is required), but it hasn't.

I guess, I need to take a look at the data.
Can you send me the KML/KMZ via mail to dennis.guse@alumni.tu-berlin.de

RugeljGG, 2023-08-01:
Not sure if you've noticed, I've emailed you the file.

dennisguse, 2023-08-02:
I got the CSV file. 
The problem is a SEGMENT_AUTOMATIC trackpoint that has no location as it was created from a distance-providing BLE sensor and that crashes the export GPX. 

Currently, I am working on changing the marking of idle by timeout rather than speed. By implemting this, the trackpoint types will also be refactored and likely reduced (no need to distinguish SENSORPOINT and TRACKPOINT as both are provided from sensors). 

@RugeljGG you try to export the problematic track as KML, import it again, and then export it as GPX.

commit c254a6ec (2023-08-03): Bug: GPXExporter - TrackPoint of type SEGMENT_START_AUTOMATIC might have no location.

Fixes #1647.