OpenTracksApp/OpenTracks #1653: GPS won't activate, when tracking starts while screen is off

v-schaefer, 2023-08-04:
I use OpenTracks in combination with Gadgedbridge and a Fossil HR watch.
Whenever I start a training on my watch (for example cycling or running) and my phone is in standby, OpenTracks doesn't activate GPS.
The message title in the statusbar stays stuck at "starting GPS" and my training will get recorded without GPS. The phone also doesn't show the symbol that an app is accessing GPS.
However if my phone is on (for example at the homescreen) and I start a new training from my watch, then GPS gets started immediately, the phone shows that GPS is getting accessed by an app and everything works fine.

All battery optimization options on Android are disabled for OpenTracks.

...

**Checklist**

- [x] Consulted [https://dontkillmyapp.com/]
- [x] Power management: battery optimization is disabled for OpenTracks

**Technical information**

- Device: OnePlus 7T Pro
- OS: Stock Android 12 ROM
- OpenTracks version: v4.6.2 (from fdroid)
- OpenTracks commit id (only for nightly builds): -

v-schaefer, 2023-08-06:
I've recorded a logcat and from what it looks like, it is no problem to create a foreground service while in standby, but the access to GPS is denied then:
[logcat.txt](https://github.com/OpenTracksApp/OpenTracks/files/12270121/logcat.txt)

To better understand what I'm doing, I've also recorded it with a second phone.
It's a bit blurry, but you can still spot what's relevant:

First Situation: Phone is on the homescreen - when the training starts, you can see the GPS-Icon on the right side of the status bar and GPS gets locked quickly:

https://github.com/OpenTracksApp/OpenTracks/assets/26538718/b80d3353-7ba2-4461-a0ff-f0c03e4a10c6

Second: The phone is in standby when the training starts. After powering on, you can see that there's no GPS-Icon in the status bar and OpenTracks is stuck on "starting GPS":

https://github.com/OpenTracksApp/OpenTracks/assets/26538718/07f76ec5-2d43-4ae3-8dc8-b08ef97974b5

dennisguse, 2023-08-06:
Location access denied.

Are you able to add (and request) the location background permission?

```
2023-08-06 19:13:10.905 10566-10566 GPSManager              de.dennisguse.opentracks             E  Not started.
2023-08-06 19:13:10.907  2595-4182  OplusLbsRepairer        system_server                        V  location access denied
```

v-schaefer, 2023-08-06:
I've tested it with the ACCESS_BACKGROUND_LOCATION permission and now it works fine!
To make things easier, I've created a PR with my modifications: https://github.com/OpenTracksApp/OpenTracks/pull/1658
Thanks for your help in pointing that out so quickly!

v-schaefer, 2023-08-08:
Fixed with https://github.com/OpenTracksApp/OpenTracks/commit/3910837d1aa3308738351c91208fa3b0d7ec1710

dennisguse, 2023-08-22:
Sadly, requesting ACCESS_BACKGROUND_LOCATION is broken.
Removing this change for now.

Likely the actual issue is: https://stackoverflow.com/questions/66475027/activityresultlauncher-with-requestmultiplepermissions-contract-doesnt-show-per

v-schaefer, 2023-08-22:
Sorry to hear that.
We only need the location in background, if the public API is activated, right?
Maybe we should only ask for that permission, if the public API is enabled.

On a fresh install the app would then ask for all permissions except ACCESS_BACKGROUND_LOCATION and as soon as the public api gets enabled, it would only ask for that additional permission.
Just to be safe and make it less hacky, we could check, what permissions we will ask for and only ask for ACCESS_BACKGROUND_LOCATION, if that's the only permission we will ask for (because all others are already granted).

What do you think?

dennisguse, 2023-08-22:
Sounds like a good plan.

dennisguse, 2023-08-22:
And this one becomes related; https://github.com/OpenTracksApp/OpenTracks/issues/1259

v-schaefer, 2023-09-09:
Hmm.. I guess the whole permission requester class will need a bit of refactoring to get both the background-access and permissions on demand running...
I tried to do that, but this seems way above my knowledge - android permissions are more complicated than I thought :(

dennisguse, 2023-09-09:
@v-schaefer what did you try? 
What was your plan?
And how can I help?

v-schaefer, 2023-09-19:
First I tried splitting up the permission request to first request everything except GPS background permission and after that only the GPS background permission.
I've done that by editing requestRequiredPermissions() in the TrackListActivity.
First I tried to just add a second PermissionRequester for the GPS background permission after the first one, which resulted in all permissions being requested at once and therefor no GPS background permission being requested.
After that I tried to request all permissions except GPS background and then open a dialog box that explains why the GPS background permission is needed and request that permission when the user clicks on "OK"... I think that this will be the right way to go, but my implementation didn't work... By clicking on OK the whole app crashed (with something like it's not allowed to start a second Activity instance).
I'm fairly new to android apps and I'm sure that we're on the right track with that idea, but at the same time it was just some spaghetti-code to test my idea and I don't think it would help if I would contribute that non-working code...

dennisguse, 2023-09-23:
Sounds like a good plan.
In addition, we could do this at two places: once when enabling the PublicAPI as well as when in StartRecording (aka when PublicAPI is triggered). In TrackListActivity, the background permission could be removed.

To request these permissions in the same activity: no clue how this is done as I never tried this :D
I am at the moment rather busy with things, but I can support you with reviews.

PS/ I would skip implementing UI elements for this.
Not really necessary and way too much effort.

sgofferj, 2023-10-17:
I'm seeing a similar issue (I think). I'm starting recording from Home Assistant by sending an intent through the HomeAssistant app to Tasker and Tasker sends the intent to OT when I leave the house with my dogs. I think, at least in my phone, it's a deep sleep issue. When I turn the phone on, even if I turn it off straight afterwards, the recording starts straight away. However, if I forget that, I get the notification from the Home Assistant app that I'm on a dogwalk but it takes between 30 seconds and forever for OT to fire up.

### App information
* ID: de.dennisguse.opentracks.nightly
* Version: 5483 v4.9.2-17-g8f9ba046c

### Device information
* Brand: ulefone
* Device: Armor_3
* Model: Armor_3
* Id: O11019
* Product: Armor_3

### Firmware
* SDK: 27
* Release: 8.1.0
* Incremental: 1535356309