OpenTracksApp/OpenTracks #1663: nightly crash

nutpantz, 2023-08-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: [e.g. Nexus 7 (2013)]
 - OS: [e.g. LineageOS 15.1]
 - OpenTracks version: [e.g. v3.2.4]
 - OpenTracks commit id (only for nightly builds): [e.g. 4b333bb]

dennisguse, 2023-08-10:
Is this reproducible?

And if so, can you get a logcat as there is likely another exception before that.

nutpantz, 2023-08-10:
It happens every time i try to run the latest nightly. I'll try for a log cat

dennisguse, 2023-08-10:
And what do you do exactly?

nutpantz, 2023-08-10:
Just start the app from the launcher.
Nothing else. It crashes immediately

nutpantz, 2023-08-10:
2023-08-10 12:02:11.572 1500 1759 system D CompatibilityChangeReporter : Compat change id reported: 197654537; UID 1000; state: ENABLED
2023-08-10 12:02:08.044 1500 4501 system W ActivityTaskManager : Tried to set launchTime (0) < mLastActivityLaunchTime (122677943)
2023-08-10 12:02:08.037 1500 12320 system I WindowManager : WIN DEATH: Window{d25f3e5 u0 de.dennisguse.opentracks.nightly/de.dennisguse.opentracks.TrackListActivity}
2023-08-10 12:02:07.893 1500 1734 system I ActivityManager : Start proc 7734:de.dennisguse.opentracks.nightly/u0a408 for pre-top-activity {de.dennisguse.opentracks.nightly/de.dennisguse.opentracks.introduction.IntroductionActivity}
2023-08-10 12:02:12.549 1500 1726 system V WindowManager : Unknown focus tokens, dropping reportFocusChanged
2023-08-10 12:02:07.883 1500 1726 system D CompatibilityChangeReporter : Compat change id reported: 135634846; UID 10408; state: DISABLED
2023-08-10 12:02:07.969 1500 2141 system I ActivityTaskManager : START u0 {cmp=de.dennisguse.opentracks.nightly/de.dennisguse.opentracks.TrackListActivity} from uid 10408
2023-08-10 12:02:07.864 1500 2141 system W ActivityTaskManager : Can't find TaskDisplayArea to determine support for multi window. Task id=8912 attached=false
2023-08-10 12:02:12.632 1500 1726 system V WindowManager : Unknown focus tokens, dropping reportFocusChanged
2023-08-10 12:02:06.476 1500 4501 system W NotificationService : Toast already killed. pkg=com.conena.logcat.reader ***@***.***
2023-08-10 12:02:07.976 1500 10891 system W ActivityTaskManager : Tried to set launchTime (0) < mLastActivityLaunchTime (122854126)
2023-08-10 12:02:07.872 1500 2141 system D CompatibilityChangeReporter : Compat change id reported: 174042936; UID 10408; state: DISABLED
2023-08-10 12:02:07.883 1500 1726 system D CompatibilityChangeReporter : Compat change id reported: 177438394; UID 10408; state: DISABLED
2023-08-10 12:02:08.010 7734 7769 de.dennisguse.opentracks.nightly E AndroidRuntime : FATAL EXCEPTION: ModernAsyncTask #2
Process: de.dennisguse.opentracks.nightly, PID: 7734
java.lang.RuntimeException: An error occurred while executing doInBackground()
	at androidx.loader.content.ModernAsyncTask$3.done(ModernAsyncTask.java:164)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
	at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
	at java.util.concurrent.FutureTask.run(FutureTask.java:271)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:920)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'android.database.Cursor android.database.sqlite.SQLiteDatabase.rawQueryWithFactory(android.database.sqlite.SQLiteDatabase$CursorFactory, java.lang.String, java.lang.String[], java.lang.String, android.os.CancellationSignal)' on a null object reference
	at android.database.sqlite.SQLiteQueryBuilder.query(SQLiteQueryBuilder.java:592)
	at android.database.sqlite.SQLiteQueryBuilder.query(SQLiteQueryBuilder.java:463)
	at de.dennisguse.opentracks.data.CustomContentProvider.query(CustomContentProvider.java:279)
	at de.dennisguse.opentracks.data.ShareContentProvider.query(ShareContentProvider.java:154)
	at android.content.ContentProvider.query(ContentProvider.java:1408)
	at android.content.ContentProvider.query(ContentProvider.java:1504)
	at android.content.ContentProvider$Transport.query(ContentProvider.java:272)
	at android.content.ContentResolver.query(ContentResolver.java:1230)
	at android.content.ContentResolver.query(ContentResolver.java:1155)
	at androidx.core.content.ContentResolverCompat$Api16Impl.query(ContentResolverCompat.java:119)
	at androidx.core.content.ContentResolverCompat.query(ContentResolverCompat.java:87)
	at androidx.loader.content.CursorLoader.loadInBackground(CursorLoader.java:63)
	at androidx.loader.content.CursorLoader.loadInBackground(CursorLoader.java:41)
	at androidx.loader.content.AsyncTaskLoader.onLoadInBackground(AsyncTaskLoader.java:307)
	at androidx.loader.content.AsyncTaskLoader$LoadTask.doInBackground(AsyncTaskLoader.java:60)
	at androidx.loader.content.AsyncTaskLoader$LoadTask.doInBackground(AsyncTaskLoader.java:48)
	at androidx.loader.content.ModernAsyncTask$2.call(ModernAsyncTask.java:141)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	... 3 more
2023-08-10 12:02:10.297 1500 1727 system W ActivityTaskManager : Can't find TaskDisplayArea to determine support for multi window. Task id=8913 attached=false
2023-08-10 12:02:08.052 1500 1726 system W ActivityManager : setHasOverlayUi called on unknown pid: 7734
2023-08-10 12:02:07.870 1500 2141 system D CompatibilityChangeReporter : Compat change id reported: 185004937; UID 10408; state: DISABLED
2023-08-10 12:02:12.564 1500 1732 system W UsageStatsService : Unexpected activity event reported! (com.android.launcher3/com.android.quickstep.RecentsActivity event : 23 instanceId : 176827347)
2023-08-10 12:02:08.011 1500 2141 system W ActivityTaskManager : Can't find TaskDisplayArea to determine support for multi window. Task id=8913 attached=false
2023-08-10 12:02:08.047 1500 4501 system W ActivityTaskManager : Can't find TaskDisplayArea to determine support for multi window. Task id=8912 attached=false
[... cut ...] t=false sz=2} to hidden list because adding Task{8ed9f4f #8913 type=standard A=10408:de.dennisguse.opentracks.nightly U=0 visible=true visibleRequested=false mode=fullscreen translucent=true sz=1}
2023-08-10 12:02:07.883 1500 1726 system D CompatibilityChangeReporter : Compat change id reported: 135754954; UID 10408; state: ENABLED
2023-08-10 12:02:07.871 1500 2141 system D CompatibilityChangeReporter : Compat change id reported: 181136395; UID 10408; state: DISABLED

That's all i get for a log.

dennisguse, 2023-08-10:
Crash also happens for me :/

dennisguse, 2023-08-10:
Database migration issue; there is another stracktrace.
Reason
I removed the TRACKPOINT type, but did not actually change the data 
One of those days...  :man_facepalming: 

```
21:38:39.553  I  Upgrade from 34 to 36
21:38:39.945  I  The connection pool for /data/user/0/de.dennisguse.opentracks.debug/databases/database.db has been closed but there are still 1 connections in use.  They will be closed as they are released back to the pool.
21:38:39.946  E  Unable to open database for writing.
                 android.database.sqlite.SQLiteConstraintException: CHECK constraint failed: trackpoints (code 275 SQLITE_CONSTRAINT_CHECK)
```

dennisguse, 2023-08-10:
@nutpantz thanks for using the nightlies!
@pstorch again thanks for hosting the nightly :D

commit 0a5b4665 (2023-08-10): Bugfix: database migration 36 did not update TrackPoint.type data (removal of SENSORPOINT).

Introduced in c0b5e719763776206222a4b39557b2e2981b84ed

Fixes #1663.
closed 2023-08-10 by dennisguse