OpenTracksApp/OpenTracks #1669: Automatic export to nextcloud

rkozlo, 2023-08-19:
I set automatic export to my nextcloud server after finished activity. Sometimes it stops working "Can't be saved"(or something similar). Manual export tracks working. Choosing the same destination directory and next activity may be send or not. Seems not to be determinstic.
**To Reproduce**
1. Set automatic export to remote nextcloud server
2. End your activity
3. Sometimes error appear "can't be saved" or something similar
4. In settings also error. Manual export works

**Technical information**
 - Device: Motorola G8 Power
 - OS: Android 11
 - OpenTracks version: 4.7.1(had this problem for whole time using this export)

Don't have any screenshots. Can gather some logs if there is a way.

pstorch, 2023-08-20:
> Sometimes error appear "can't be saved" or something similar

Can you give more details on "or something similar"? Exact error message might be helpful.

rkozlo, 2023-08-20:
I do not have English language set. So i translate it by myself, so English version it may sound different. Don't see a way to change language in application so it uses statically system language.

pstorch, 2023-08-27:
We've added some diagnostic to analyze this issue. It's already available in the nightly and soon in the next release. 
I hope we get some more insight.

WPettersson, 2023-09-03:
Just had the issue:
```
Post-workout export failed, please check the export directory.
Cannot write to export directory.: androidx.documentfile.provider.TreeDocumentFile@cde2130
```

Note that this was a first export after upgrade, I don't know whether perhaps the app needs to be re-authorised after an upgrade.

pstorch, 2023-09-03:
Looks like no write permission. Please try to authorize the the directory again. Let's see if it helps.

WPettersson, 2023-09-03:
Have done, just have to wait for the next error now

dennisguse, 2023-11-21:
No error reported, so this might be working ;)

closed 2023-11-21 by dennisguse