OpenTracksApp/OpenTracks #1671: excessive Average / max moving speed when starting to track / while tracking (when walking)

Der-Flip, 2023-08-20:
**Describe the bug**
Average and Max speed unrealistically high.
Either right after starting it jumps to 1000+km/h or it'll creep up over time while walking. I've been walking fairly slow, unsure if that may contribute.

**To Reproduce**
1000+kph:
1. Be stationary
2. Open app
3. Activate position retrieval
4. Start tracking
5. Start moving

Creeping up over time:
1. Open app
2. Keep walking
3. Start recording

If applicable:
[excessivemovingspeedgpx.zip](https://github.com/OpenTracksApp/OpenTracks/files/12387729/excessivemovingspeedgpx.zip)

**Technical information**
 - Device: Xiaomi Redmi9C 
 - OS: Android 10
 - OpenTracks version: v4.7.1
 - OpenTracks commit id (only for nightly builds): ---

dennisguse, 2023-08-20:
I guess this is recent; since when did you notice this?

Der-Flip, 2023-08-20:
Unfortunately I don't know. I haven't used the app in a while as I wasn't really out and about in recent times and it is just now on vacation I noticed :/

dennisguse, 2023-08-20:
Don't worry ;)
PS/ can you export one of these tracks as CSV?
The CSV exporter skips all the preprocessing required for GPX/KML and thus easier to read :D

Der-Flip, 2023-08-20:
Didn't find the single file export as CVS but I managed ^^

[2023-08-19_12_57_09.095_2023-08-19T12_57+02_Andelsbuch_Bezau_Tag5.csv](https://github.com/OpenTracksApp/OpenTracks/files/12389043/2023-08-19_12_57_09.095_2023-08-19T12_57%2B02_Andelsbuch_Bezau_Tag5.csv)
[2023-08-17_12_26_43.707_2023-08-17T12_26+02_Uga_Tag3.csv](https://github.com/OpenTracksApp/OpenTracks/files/12389077/2023-08-17_12_26_43.707_2023-08-17T12_26%2B02_Uga_Tag3.csv)
[2023-08-20_14_57_51.281_2023-08-20T14_57+02_Quelltuffstein_Tag6.csv](https://github.com/OpenTracksApp/OpenTracks/files/12389037/2023-08-20_14_57_51.281_2023-08-20T14_57%2B02_Quelltuffstein_Tag6.csv)

dennisguse, 2023-08-20:
@Der-Flip thx.
First glance at the data: looks okay.
So, we might have a bug in the `TrackStatisticsUpdater`.

Der-Flip, 2023-08-20:
So for today's I started recording, moved and it jumped to well above 1000kph in the initial fixes. Later it reduced but nothing plausible really. I don't wanna deny any mistakes I may make though.

GPS settings:
[image]

I also noticed that the current speed - despite me walking - shows zero every now and then. I suspected maybe the gps fix distance may be too long for my walking speed in this case. Reducing it to the 2m seen above I felt like it made somewhat of a difference but I think that was more of a placebo.

Der-Flip, 2023-08-30:
Past days were a bit stressful and hectic so I couldn't update you any sooner:

So after the update I did a bike tour and I thought it was fixed.

The day after I did another hiking tour and I realized it wasn't.

It seems as if the current speed jumps to 0 as soon as my speed drops to around 3.5ish km/h. While walking 4kph it worked fine. If my walking speed dropped (mountains, they are cruel) it showed 0 despite me moving at ~2km/h. Once I moved a substantial distance (anything between 10-20m) it updated with a slightly elevated current speed and displaying an unreasonably high max speed. Sometimes it dismissed these unreasonable values sometimes it kept them.

I checked with my fitness tracker (integrated, independent gps) as well as phone's gps and I was definitely above the 0.5km/h cutoff.

dennisguse, 2023-08-30:
Could you check with the current nightly?
It already contains the #1652 by using an idle timeout rather than using the speed as a threshold.

closed 2024-12-19 by dennisguse