OpenTracksApp/OpenTracks #1673: Extreme gain and loss values

kaibagley, 2023-08-21:
**Describe the bug**
Stats for runs shows the gain and loss to be very high (e.g. gain of 6018 m and loss of 6021 m).

When viewing the GPX on Nextcloud app GPXPod, the gain and loss are much more believable (~ 20 m)

**To Reproduce**
1. Start recording
2. Go on a run
3. Stop recording
4. View stats

**Technical information**
 - Device: Pixel 7 Pro
 - OS: GrapheoeOS
 - OpenTracks version: 4.6.2

dennisguse, 2023-08-22:
Please test your device with the following procedure: https://github.com/OpenTracksApp/OpenTracks/blob/main/README_TESTED_SENSORS.md#tested-barometric-sensor-smartphones

closed 2023-08-22 by dennisguse

kaibagley, 2023-08-22:
Phone is a 1 month old Pixel 7 Pro running GrapheneOS. Doing those test, phone passes all 4 tests, with test 4 passed by going up and down my stairs and got + or - 3 m each time...

Starting to think that I changed a system setting somewhere or something, will give an update after another run soon