OpenTracksApp/OpenTracks #1676: App crashes when saving a traxk

elumbella, 2023-08-26:
App just crashed when saving a very short track. Auto export (GPX) is enabled.

### App information
* ID: de.dennisguse.opentracks
* Version: 5361 v4.8.0

### Device information
* Brand: Sony
* Device: XQ-CC54
* Model: XQ-CC54
* Id: 65.1.A.5.65
* Product: XQ-CC54_EEA

### Firmware
* SDK: 33
* Release: 13
* Incremental: 00650001A005006503491578438

### Cause of error
```java
Exception in thread "main": java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.services.TrackRecordingServiceNotificationManager.updateContent(java.lang.String)' on a null object reference
	at de.dennisguse.opentracks.services.TrackRecordingService.newGpsStatus(TrackRecordingService.java:247)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.sendGpsStatus(TrackPointCreator.java:151)
	at de.dennisguse.opentracks.services.handlers.GPSManager.onGpsStatusChanged(GPSManager.java:169)
	at de.dennisguse.opentracks.services.handlers.GpsStatusManager.setGpsStatus(GpsStatusManager.java:165)
	at de.dennisguse.opentracks.services.handlers.GpsStatusManager.determineGpsStatusOnTrackpoint(GpsStatusManager.java:111)
	at de.dennisguse.opentracks.services.handlers.GpsStatusManager.onNewTrackPoint(GpsStatusManager.java:91)
	at de.dennisguse.opentracks.services.handlers.GPSManager.onLocationChanged(GPSManager.java:117)
	at androidx.core.location.LocationListenerCompat.onLocationChanged(LocationListenerCompat.java:46)
	at android.location.LocationManager$LocationListenerTransport$1.operate(LocationManager.java:3252)
	at android.location.LocationManager$LocationListenerTransport$1.operate(LocationManager.java:3249)
	at com.android.internal.listeners.ListenerExecutor.lambda$executeSafely$0(ListenerExecutor.java:127)
	at com.android.internal.listeners.ListenerExecutor$$ExternalSyntheticLambda0.run(Unknown Source:8)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:346)
	at android.os.Looper.loop(Looper.java:475)
	at android.app.ActivityThread.main(ActivityThread.java:7950)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:942)
```

dennisguse, 2023-08-26:
Thanks for the report.
This is fixed in v4.8.1 (Not yet on Fdroid).

More details in #1668

closed 2023-08-26 by dennisguse

dennisguse, 2023-08-26:
PS/ the nightly already contains this fix.
And it is not the export that is broken: the problem is related to stopping a recording.
Manual export works.