OpenTracksApp/OpenTracks #1680: GPS not starting

PeteMacca, 2023-08-28:
**Please describe the problem and how to reproduce it:**

If Location permission is set to "Allow only when using the app", GPS does NOT start.

If Location permission is set to "Allow all the time", GPS does start.

It used work fine with the "Allow only..." setting prior to this latest version.
...

**Checklist**

- [x] Consulted [https://dontkillmyapp.com/]
- [x] Power management: battery optimization is disabled for OpenTracks

**Technical information**

- Device: [Pixel 3a]
- OS: [Android 12]
- OpenTracks version: [v4.8.0]
- OpenTracks commit id (only for nightly builds): [5361]

dennisguse, 2023-08-28:
Sadly, this is a bug in v4.8.0 (see #1668);
Please upgrade to v4.8.1 (F-Droid publication still pending).
As an intermediate solution: the nightly works.

closed 2023-08-28 by dennisguse

PeteMacca, 2023-08-29:
Cheers, mate!