OpenTracksApp/OpenTracks #1681: Start of the week - selectable

natrius, 2023-08-28:
**Is your feature request related to a problem? Please describe.**
For almost half of the worlds population the start of the week is Monday and not Sunday. https://www.timeanddate.com/calendar/days/first-day-of-the-week.html 
Also https://en.wikipedia.org/wiki/Week

Its incredible annoying when filtering when this setting is missed. 

**Describe the solution you'd like**
Settings - User Interface - "Start of the week"
Let us choose between Saturday, Sunday and Monday

**Describe alternatives you've considered**
None

dennisguse, 2023-08-28:
Actually, OpenTracks behaves region specific.
With English (US) it is Sunday.
With German (Germany) as well as  Spanish (Spain) it is Monday.
So, works as expected :D

Android 14 will support configuring certain settings per Language/Region.
https://developer.android.com/about/versions/14/features#regional-preferences

natrius, 2023-08-31:
I had region UK and it was sunday. Changed the region now to austria and its still sunday. Changed the regional settings to germany, its still sunday. 

Every calendar app has included a setting for this, please open this issue again and it would be greatly appreciated if this would be a setting.

pstorch, 2023-08-31:
[image]
[image]

pstorch, 2023-09-02:
Tested again. The system setting which I showed above applies only to the stock calendar app, no other app respect that setting.
But when I change the language from english/us to english/uk, I get the starting date of monday.

dennisguse, 2023-09-23:
Here is the code that is determining the start of the week: https://github.com/OpenTracksApp/OpenTracks/commit/b4ce99440a364485c5f2e08e238efefe7c9c3e14

It is not Android 14 ready :)

dennisguse, 2023-10-12:
We may also think about migrating to MaterialDesign's DateRangePicker

PR #1984 (2024-09-08): Aggregated stats: start of the week uses regional preferences (Android14+).
Implemented using Material3's DatePicker.

Fixes #1681.
commit 79254550 (2024-09-08): Aggregated stats: start of the week uses regional preferences (Android14+).

Implemented using Material3's DatePicker.

Fixes #1681.
closed 2024-09-08 by dennisguse