OpenTracksApp/OpenTracks #1682: Crash when a track is shared

githubcatw, 2023-08-30:
**Describe the bug**
When the Share button is pressed on a track the app crashes.

**To Reproduce**
1. Go to any recorded track.
2. Tap the Share button.

**Technical information**
 - Device: Google Pixel 6
 - OS: Android 14 beta 5.2
 - OpenTracks version: 4.8.1

githubcatw, 2023-08-30:
Here's the log that OpenTracks shows:

```java
Exception in thread "main": java.lang.IllegalArgumentException: UUID string too large
	at java.util.UUID.fromStringJava11(UUID.java:238)
	at java.util.UUID.fromString(UUID.java:226)
	at de.dennisguse.opentracks.io.file.TrackFilenameGenerator.getExample(TrackFilenameGenerator.java:96)
	at de.dennisguse.opentracks.io.file.TrackFilenameGenerator.isValid(TrackFilenameGenerator.java:86)
	at de.dennisguse.opentracks.settings.PreferencesUtils.getTrackFileformatGenerator(PreferencesUtils.java:659)
	at de.dennisguse.opentracks.share.ShareUtils.newShareFileIntent(ShareUtils.java:58)
	at de.dennisguse.opentracks.TrackRecordedActivity.onOptionsItemSelected(TrackRecordedActivity.java:174)
	at android.app.Activity.onMenuItemSelected(Activity.java:4525)
	at androidx.activity.ComponentActivity.onMenuItemSelected(ComponentActivity.java:529)
	at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:352)
	at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:269)
	at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:110)
	at androidx.appcompat.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:66)
	at androidx.appcompat.widget.Toolbar$1.onMenuItemClick(Toolbar.java:225)
	at androidx.appcompat.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:781)
	at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:836)
	at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:159)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:987)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:977)
	at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:625)
	at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:156)
	at android.view.View.performClick(View.java:7659)
	at android.view.View.performClickInternal(View.java:7636)
	at android.view.View.-$$Nest$mperformClickInternal(Unknown Source:0)
	at android.view.View$PerformClick.run(View.java:30156)
	at android.os.Handler.handleCallback(Handler.java:958)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:294)
	at android.app.ActivityThread.main(ActivityThread.java:8177)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
```

dennisguse, 2023-08-30:
1. Does this really happen for all tracks?
2. .. if so can you record a new and try again?
3. Does exporting (via settings) work?

Why I am asking this:
For each track a UUID should be generated when the recording started
For imports: create a new one if none is the KML/GPX.
... so what you found shouldn't happen :D

githubcatw, 2023-08-31:
> 1. Does this really happen for all tracks?
> 2. .. if so can you record a new and try again?
> 3. Does exporting (via settings) work?

1. At least for all the tracks that I have.
2. Recording a new track results in the same error.
3. Tapping "Import/export options" results in the same error too.

> Why I am asking this:
> For each track a UUID should be generated when the recording started
> For imports: create a new one if none is the KML/GPX.
> ... so what you found shouldn't happen :D

The error is about a UUID being too large, so I think it's being incorrectly created, but I can't look into it right now.

Also, viewing it on a map as KML or GPX works. Samsung Health doesn't accept the GPX files but [GPX Viewer](https://play.google.com/store/apps/details?id=com.vecturagames.android.app.gpxviewer) (it's an app on Google Play) does.

dennisguse, 2023-08-31:
Mhhh.
The problem occurs here: https://github.com/OpenTracksApp/OpenTracks/blob/bb722de756c27f0f687319a6386ee19fa6f1f0c0/src/main/java/de/dennisguse/opentracks/io/file/TrackFilenameGenerator.java#L93

```
        track.setUuid(UUID.fromString("fefefefefe-0000-1000-8000-00805f9b34fb"));
```

And there we parse a UUID from a predefined string (not even the database) to show the filename as an example. And that works since 3 years...
And `UUID.fromString()` is ancient: https://developer.android.com/reference/java/util/UUID#fromString(java.lang.String)

dennisguse, 2023-08-31:
Can confirm this; this will occur from Android 14+ due to Java11 alignments.
Reading source code helps...

```
    public static UUID fromString(String name) {
        // BEGIN Android-changed: Java 8 behaviour is more lenient and the new implementation
        // might break apps (b/254278943).
        // Using old implementation for apps targeting Android older than U.
        if (!(VMRuntime.getSdkVersion() >= VersionCodes.UPSIDE_DOWN_CAKE
                && Compatibility.isChangeEnabled(ENABLE_STRICT_VALIDATION))) {
            return fromStringJava8(name);
        }

        return fromStringJava11(name);
        // END Android-changed: Java 8 behaviour is more lenient and the new implementation
        // might break apps (b/254278943).
    }
```

commit 4ca6b230 (2023-08-31): Bugfix: prevent crash due to invalid UUID example (Android 14+ / API34+).

Fixes #1682.
closed 2023-08-31 by dennisguse